andrivet/ADVi3pp #4: [FR] BLTouch

andrivet, 2017-10-24:
### Feature Request

Give instructions to install [BLTouch](https://www.antclabs.com/bltouch) and enable it in the firmware. There are apparently some room (i.e. enough pins) on the i3 Plus motherboard.

[image]

sniperxnl, 2017-11-15:
That would be nice to add to the I3+, is there are workout how to connect it to the mainboard already?

weidercs, 2017-11-19:
Do you want me to add some instructions here or will you just take it from my fork?

andrivet, 2017-11-19:
I will take the instructions (and configuration) from your fork.

alexten9, 2018-03-29:
weidercs i took your files loaded to printer and when I do G28 Z just moves up and all others go home

alexten9, 2018-03-29:
weidercs i was able to use your files can you point to the next step on how to tune it in.

andrivet, 2018-04-02:
Implemented in release 3.0.0

closed 2018-04-02 by andrivet