andrivet/ADVi3pp #8: [FR] Make changing filaments during a print easier.

brookpatten, 2017-10-24:
I'd really like to see the ability to pause a print and swap filaments and have it return to where it was in the print to resume.  If it could cover both the "print from sd" and the "print from pc/octoprint" scenarios then all the better.  It could be implemented as a more general function of just "remember this x/y/z" "return to remembered x/y/z" as well.

andrivet, 2017-10-24:
Do you mean: implementing a working M600 gcode? Yes, it would be great!

brookpatten, 2017-10-24:
It sounds like that would do it yes.  Sorry I'm not familiar with much gcode.

Even it was just a UI button where I could pause, move, unload, load, "return", and resume that would do the trick too for a lot of cases.  I think having m600 where you could plan a change is great too, but I was thinking more along the lines of "oh shoot I'm almost out of filament" type scenarios where the gcode is already running and you want to do it from the UI.

Burritobun, 2017-11-03:
Looks like the M600 code really would need an LCD to let you know what steps you have to take next. Getting the LCD to show M117 codes would be the first thing to sort out.

chschmit, 2017-11-23:
The Wanhao3 firmware allows pausing the printer to make a filament changing during printing. When you press the UI pause button, the head immediately moves to front of the print bed and the filament can then be manually inserted (auto filament loading doesn't work). After you've 'primed' the nozzle with the new filament, pressing the UI play button will resume the print job. I've made many flawless prints this way, especially when I'm trying to use up short pieces of filament.
This is one of the few features I like about the Wanhao3 firmware, that the alternate ones seem to miss.

Scottm80, 2017-12-20:
I am also having issues with this.

I use Octoprint and S3d. I want to colour swap on prints but I can't. The machine seems to loose home position. 
I have tried using multiple processes in S3d and the new multi color plugin with Octoprint. Both methods will not work because the firmware looses the home position. So frustrating.

Mobsbobs, 2018-01-03:
This process for colour change does work in S3D. you create multiple processes and in "Prepare To Print" you save each process as a seperate print job. I have done many multi colour prints this way.

sniperxnl, 2018-01-03:
M600 code is the best way to use what i have tested last year, thats why i
asked

sniperxnl, 2018-01-20:
@Mobsbobs can you show me how?

andrivet, 2018-04-29:
Implemented in Beta 3

closed 2018-04-29 by andrivet