andrivet/ADVi3pp #13: [FR] Auto bed leveling enhancement

andrivet, 2017-11-08:
### Feature Request

From Johannes Schoeller:

As auto bed leveling now works: add some quick adjustment of the nozzle height (in relation to the sensor), move up/down by eg. 0.05mm 
-> M851 = read offset
-> add/subtract 0.05 to that value
-> M851 Z[new value]
-> M500
something like that. this can make adjustements easier w/o needing to note those codes and w/o a PC.

iz3man, 2017-11-30:
i may some how to maybe:
playing with the serial console and changing M851 up and down, then saving, doing it again etc takes some time. sure this is something you may do once, and then forget.
but if you tinker a lot and change your printbed, your nozzle etc you have to do it all again.
so a menu like the "move" menu where you can do the following:

.) move nozzle down to the bed (Z0)
.) then test with a sheet of paper (or your eyes, or whatever)
.) have a button for 1mm up/down, 0.1mm up/down, 0.05mm up/down
.) have a button for save (M500)

that's it :)
thank you

andrivet, 2019-05-26:
No more relevant with version 4.0

closed 2019-05-26 by andrivet