andrivet/ADVi3pp #15: Bug - Will not y-home if you cancel a print and then restart

rippledabs, 2017-11-11:
### Bug Report
- Description: If you cancel a print and then restart it. It will not y-home. 
- Expected behaviour: y-home
- Actual behaviour: It just uses the current position as the y-home for the next print. 
- Steps to reproduce:
  - Cancel a print
  - Restart the print

andrivet, 2017-11-19:
I am able to reproduce this bug in the case of a print from the SD card. Is it correct or do you also see this bug during an USB print?

andrivet, 2017-12-09:
From my latest tests, the bug is still there

Arsoth, 2018-01-02:
I can confirm this bug is still occurring on 2.0.1 through USB (at least when using Cura 3.1.0 to print).

TTVert, 2018-02-12:
FYI, this occurred for me w/ stock firmware using repetier host as well.

Dave

TTVert, 2018-02-24:
Curious if this will be something that can be resolved?  This also occurs if you queue up a job in repetier-host before printer is on.  It will start right where the printhead was last left.  It does home Z however.

andrivet, 2018-02-24:
The fix is planned for version 3.0.0

TTVert, 2018-02-24:
I see you had screenshots of screen brightness, how close is 3.0?

Martin-Dolan, 2018-04-01:
This is probably the same thing, but I was doing a print in ABS and just laying down the first layer. I realised it was going to turn to crap as the bed wasn't level enough so I stopped the print, went to the bed level menu, started the level option however it seemed that it thought the x axis "0" was the point on the bed where I had pressed stop.

TTVert, 2018-04-01:
Yep exact situation I run into often.  I have to turn printer completely off and manually home w/ G28  Reset doesn't fix for me.

Dave

Martin-Dolan, 2018-04-01:
I did notice after I had turned it off and on, when I went and started the level process again I saw a screen that said something about "waiting for homing to complete" or something like that. It was only there for a few seconds.  I am 99% certain that it did *NOT* do that when I did the level process after aborting a print so that might help narrow it down?  It is skipping that homing process part of the code when a print has been aborted.

TTVert, 2018-04-01:
It's an issue w/ the marlin FW.  Stock also.  andrivet said he was going to resolve it in 3.0.

Dave

SureshotM6, 2018-04-01:
The workaround I have been using so far (which I found on another forum and did not discover myself) is to disable steppers with M18 before homing.  Seems to be working so far, easier than power cycling everything, and is easy enough to put into the generated gcode or octoprint.

SureshotM6, 2018-04-09:
> Is someone able to reproduce this bug in a reproducible way?

Yes.  A M17 alone should trigger it (at least on my printer).  For example, run the following after moving the extruder somewhere in the middle of the bed where the homing issue is more easily visible:

```
M17 ; enable all steppers including the extruder
G28 X ; Try to home X.  Won't work
```

I did some measurements and it appears to be a hardware issue.  I've been using this code I wrote as a fix: 5da26d65 "i3plus: fix X homing when extruder is enabled."  There are more details in the commit message, but basically, there is noise from the extruder stepper on the X_MIN pin that is causing the AVR to detect the limit switch as closed when it is actually open.  Feel free to cherry pick that if you agree with the fix.

andrivet, 2018-04-10:
@SureshotM6 I confirm that M17 / G28 X does exhibit the problem. I will try to see the signals with an oscilloscope

andrivet, 2018-04-11:
Fixed in Beta 3

closed 2018-04-11 by andrivet

TTVert, 2018-04-11:
Do you have a beta tester area for your FW?   I know you said 3.0 was near and some functionality of it I can use.

Dave

andrivet, 2018-04-12:
@TTVert Sorry, I don't understand what you are asking.

andrivet, 2018-04-13:
Yes of course I need beta testers.

ninjanickk, 2018-04-27:
any idea when beta 3 will be out. i have the issue with my x axis always being triggered and im really hoping beta 3 will help. thank you for your hard work.

TTVert, 2018-04-28:
Why are you deleting our comments?

andrivet, 2018-04-28:
Because this is not the right place to ask for support. For support, please ask your question on Facebook group or on Patreon. I try to keep comments on issues "clean", i.e. with only (technical) information about the issues. So I will also delete this comment.

Mechotronic, 2018-07-11:
Just wanted to say thanks for this! I wonder if a stronger pullup would fix this on the daughter board?

commit e2c6fb37 (2018-04-11): Fix #15: Will not x-home if you cancel a print and then restart. Thanks to Matt Cowell. https://github.com/andrivet/ADVi3pp-Marlin/commit/5da26d65fd23c923a3d7c16d39c7518396392a60