andrivet/ADVi3pp #16: Bug - If there is a problem reading the flash drive, the screen blacks out

rippledabs, 2017-11-11:
------------------------------------------------------------
### Bug Report
- Description: If there is a problem reading the flash drive, the screen goes into the temperature plot screen. If you attempt to go back, the screen blacks out and you have to reboot. 
- Expected behaviour: Some error message about the flash drive.
- Actual behaviour: Everything locks up. 
- Steps to reproduce:
  - Insert flash drive
  - Press the print button
  - Instead of going to the flash directory, it goes to the temperature plots. 
  - Press back
  - Screen blacks out.

andrivet, 2017-11-13:
If there is no SD card, the print button displays temperatures. This is expected. For the other part (blacks out), I will check that.

andrivet, 2017-11-19:
It is difficult to test such scenario (bad SD, ...) but I think that the black out is solved. However, Marlin does not handle well some cases (such as removing the SD card during a print) and I will not fix such corner cases.

closed 2017-11-19 by andrivet