andrivet/ADVi3pp #19: Calibrating issue

sniperxnl, 2017-11-15:
Guys,

I have loaded today the firmware, and as always busy to calibrate the multiplier. Printing and messure. Now 4 walls one print and all diffrent thickness, how is that possible? 

[image]
[image]

andrivet, 2017-11-15:
I just performed a test myself (25x25x25 mm cube, height 0.12, wall thikness 1, no top/bottom layer) with Cura 3 and I have consistant measures for the 4 walls.
Which slicer are you using, with which parameters?

sniperxnl, 2017-11-16:
I am using s3d, 2 walls print on 0.2 with a 0.40 nozzle. Cube is 20x20x15 and print with tpu

sniperxnl, 2017-11-16:
solved

closed 2017-11-16 by sniperxnl