andrivet/ADVi3pp #24: Compilation issues with Arduino IDE

andrivet, 2017-11-18:
### Bug Report
- Description: It is not possible to compile version 2.0.0 Beta 1 with Arduino IDE
- Expected behaviour: Compilation working
- Actual behaviour: Errors such as:
```
ccVQHPBh.ltrans0.o:(.text+0xe32): undefined reference to `advi3pp::LCD::has_status()'
ccVQHPBh.ltrans0.o:(.text+0xe40): undefined reference to `advi3pp::LCD::set_status_PGM(char const*, signed char)'
ccVQHPBh.ltrans0.o:(.text+0x10a4): undefined reference to `advi3pp::LCD::refresh()'
ccVQHPBh.ltrans0.o:(.text+0x1be2): undefined reference to `advi3pp::LCD::set_status_PGM(char const*, signed char)'
```
- Steps to reproduce:
  - Open the project with Arduino IDE
  - Verify

Bosbessenbasje, 2017-11-18:
I second this. As I have a different thermocouple arrangement, I need to fine tune the marlin software and recompile. I got the same errors as Andrivet reported.

andrivet, 2017-11-18:
Temporary workaround: If you take the current content of the repository (https://github.com/andrivet/ADVi3pp-Marlin/archive/advi3++.zip), it compiles. Another workaround is to use a more recent version of Arduino IDE.

andrivet, 2017-11-19:
I now compiles with Arduino IDE 1.8.5 or 1.9 (Beta)

closed 2017-11-19 by andrivet