andrivet/ADVi3pp #26: Y missing steps

sniperxnl, 2017-11-18:
Guys, I have done serval prints on different location on the bed, and the y axis have a missing step always on the same height,. I have already checked all the belts. The bed is smooth no cable is stuck. Looks like it happend after this new firmware

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

Multiplxd, 2017-11-19:
I flashed my machine with 1.0,1 and am now having the same issue with either jitter or missing steps on the Y axis

closed 2017-11-19 by andrivet

sniperxnl, 2017-11-19:
Why is it closed? Any solution?

andrivet, 2017-11-19:
Because if you flash another (stock) firmware and still have the same problem, it is not related to the firmware.

sniperxnl, 2017-11-19:
I have it only on the marlin firmware

Multiplxd, 2017-11-19:
I flashed over from the latest wanhao where I didn't have this issue.

sniperxnl, 2017-11-21:
Hai @andrivet  found the issue, my new SD card was the issue.

andrivet, 2017-11-21:
Happy to learn that!