andrivet/ADVi3pp #27: [FR] Progress bar

chschmit, 2017-11-18:
A progress bar would be great that could be viewed across a room. I use multiple printers and the progress bar on the Wanhao latest firmware worked very well from a visibility standpoint.

TTVert, 2018-02-12:
I would also like this.  Perhaps put the bar graph where the pause/play/stop icons were before in graph?

mpfl, 2018-02-24:
This would work very well with the M73 Progress plugin for OctoPrint.

andrivet, 2018-07-04:
It may look like this:

[image]

[image]

TTVert, 2018-07-04:
Is the green the portion still to be completed?  If so would it be better to have a white or red box w/ teh green showing what is completed?  
[image]

chschmit, 2018-07-05:
I agree with this revised approach. Conventional progress bars show completeness from left to right generally with a bolder color signifying the amount complete. White, gray or transparent are usually the portion of the bar that isn't completed yet.

andrivet, 2018-07-05:
OK, so it may looks like this:

[image]

[image]

TTVert, 2018-07-05:
Much better!

chschmit, 2018-07-05:
Agreed. Simple, yet very usable.

Deose, 2018-08-02:
I think a ETA would be more important than a %, or a time elapsed + a % completed.

andrivet, 2018-08-03:
@Deose "I think..." Why do you think it is better? And computing ETA is a very complex problem. How do you see that?

TTVert, 2018-08-03:
Could you not use the ETE/ETA reported from the slicer?

andrivet, 2018-08-03:
@TTVert Outputting ETA/ETE on the LCD is already possible with the M117 support (there are already OctoPrint's plugins for that). Regarding G-Codes, there is [M73](https://reprap.org/wiki/G-code#M73:_Set_build_percentage) but it is not supported by Marlin. I am not aware of any G-Code with ETA. And even if the slicer gives ETA information, it will not be accurate and I will then get many bug reports saying the ETA is wrong (this is what happened with OctoPrint). And I want to avoid such situation. 

So if someone has a reliable (and documented) way of giving a reliable ETA, I may implement this. For the moment, it is no.

TTVert, 2018-08-03:
Personally I'm fine w/ percentage and using repetier I know first hand how inaccurate ETA/ETE can be.

churd83, 2018-08-03:
Personally as long as it completes I'm good

Deose, 2018-08-04:
@andrivet I think an ETA is better than a progress bar or a % because isn't the whole purpose of showing the % completed or a progress bar to show "How close it is to being completed"? If the purpose of said things are to see how far along it is, a straight ETA not old tells that as well, but gives extra information as well.

I know absolutely nothing about implementing these things or coding it, I just saw people commenting on the markups and wanted to throw my 2¢ out there.

Like somebody else said, Cura has a "Total time estimated" and "time estimated remaining" when it prints. Even the stock Maker Select plus has a time gauge on it, I believe. I don't fully remember now (as I always used Print from PC via USB), but think it has some sort of count down (or count up) function built it.

Freecoding an ETA function by hand could be difficult, and like you said, might be a lot of work if you don't want to deal with people submitting bugs about it. However, a % of completed (I guess it could divide how many lines it has down, divide it by how many lines there are total, and multiply it by 100) and a running timer from when it starts printing, together would make it so people could estimate how much longer. It would be quite a rough estimate, since different layers are different sizes, but it's at least some way to gauge how long a print has left.

chschmit, 2018-08-04:
As the originator of this request I also desire an ETA status but due to the inaccuracies of any ETA calculation I've seen so far I was hoping to have a progress bar that can be viewed from across the room similar to the Wanhao original firmware.

Let's start with a progress bar and if there is an ETA implementation that actually works in the future I'd support that as text overlaying a progress bar.

therevoman, 2018-08-04:
I would love to see a semi accurate ETA but understand how difficult that problem is.  Have you ever watched the progress bar of a large file transfer oon your computer? The ETA changes regularly.  That is because they make guesses on transfer speed in the beginning and then use a rolling average transfer rate later. So much math and guessing.

I really like knowing how many layers remain and would appreciate a progress bar based on number of layers.

For the original poster, just think of the different settings your slicer gives you depending on the layer being printed, along with the differences in a defined speeds and actual speeds...  well, it goes way beyond that. 

As much as I'd like to see an ETA progress bar, I'm with the developer on avoiding it's addition.

TTVert, 2018-08-04:
repetier also shows layers, it cycled through that, eta, etc.

andrivet, 2018-08-24:
Implement as the same time M73 - LCD_SET_PROGRESS_MANUALLY

andrivet, 2018-08-24:
Implement also M250 - LCD Contrast

closed 2018-11-04 by andrivet