andrivet/ADVi3pp #36: Long filenames displace percent complete status

chschmit, 2017-11-23:
------------------------------------------------------------
- Description: Filenames that are too long prevent the percent complete status from showing
- Expected behaviour: The percent complete is visible regardless of the length of the filename.
- Actual behaviour: The filename likely forces the percent complete status out of view.
- Steps to reproduce:
  - While printing the filename and percent complete status are normally visible on the temperature graph, if the filename is too long, the percent complete status is missing.

Martin-Dolan, 2018-03-25:
Haha I was just about to ask if there *was* a percent complete display. I guess all my file names are way too long because I've never seen it!

andrivet, 2018-04-07:
I am not able to reproduce. Can you give an example? "Filenames that are too long" is a little bit vague. Close for the moment. Do not hesitate to reopen if you add more details

closed 2018-04-07 by andrivet

chschmit, 2018-04-07:
As of 3.0 Beta 1, the percentage status is visible AND I'm using longer filenames (at least 30 characters long). Previously if the filename and extension had more than 18 characters the percentage status didn't show up. Thank you.

jluzonpr, 2018-05-02:
@andrivet I am right now printing `ZeroOffset_Bracket.gcode` in 2.1.0 and only one digit of the progress percentage. Seems to me like any more digits after that could potentially make it so the progress number just doesn't show. 

I understand from what @chschmit posted this might be fixed or harder to run into in 3.0 Beta updates, but I haven't tried those yet.