andrivet/ADVi3pp #37: Some times when starting a new print the extruder removes the filament

andrivet, 2017-11-25:
This is not yet clear if it is a bug in the firmware or not. However, from some tests, I think so.

Arsoth, 2018-01-02:
I've just had this occur twice while attempting prints - at between 144 and 154mm of filament extruded, the extruder retracts all of the filament out of the hot end, and past the extractor gear itself. 

It occurs at roughly the same point; both times during layer 4 of my calibration "cube", both times with separate slicer settings used to generate the G-code. Reprinting with the exact same  G-code did not cause the unexpected retraction when I tried again.

Attached is the Gcode where it occurred. It was during layer 4; if the code was to be printed and/or visualized, the retraction happened while almost done with the innermost wall along the curved corner.
[Calibration "cube" gcode.txt](https://github.com/andrivet/ADVi3pp-Marlin/files/1596805/Calibration.cube.gcode.txt)

andrivet, 2018-04-22:
Issue is still there. From my tests, it is related to bug #97

andrivet, 2018-04-22:
Fixed in Beta 3

closed 2018-04-22 by andrivet