andrivet/ADVi3pp #42: [FR] Fillament runout sensor

sniperxnl, 2017-11-27:
Is it possible to add that?

andrivet, 2017-11-28:
Please be more precise. Which runout sensor? To evaluate if I can add something into the firmware, I have to know what.

jonathanweinberg, 2017-12-12:
Usually a filament runout sensor is just a NC (normally closed) switch. Here is an example: https://www.thingiverse.com/thing:2665162

When it goes open, Marlin registers the filament has run out and pauses the print. Scroll down to “Filament Runout Sensor”: http://marlinfw.org/docs/configuration/configuration.html

andrivet, 2017-12-13:
IBow do you plan to connect it to the motherboard?

jonathanweinberg, 2017-12-13:
Sorry, I was just passing through and noticed a question I could help clarify. Don’t even own a Plus.

andrivet, 2017-12-13:
Closed because no (useful) feedback or information

closed 2017-12-13 by andrivet