andrivet/ADVi3pp #44: Thermal Runaway Errors

andrivet, 2017-12-02:
Please comment this issue to report problems with Thermal Runaway Errors. Please provide:
- Are you printing from the SD card or using USB?
- In the case of USB, with which software (name and version)?
- What are your target temperatures (both Bed and Hotend)?
- Does it occurred immediately, after a while, randomly? Please provide as much information as possible.

andrivet, 2017-12-02:
From Robin Cleland:
I'm printing directly from Simplify3D 4.0

I'm on ADV 2.0.1

I'm getting the thermal runaway error with the extruder at about 27 degrees and the bed at about 35 degrees being heated to 60 degrees.

Jannibal, 2017-12-17:
Printing from Octoprint 1.3.6 over USB
Trying to print a temperature tower that starts at 260C and goes down to 230C.
When the hot end reaches 260C before starting to print (which is the temperature to pre-heat to) I get this message, and need to reconnect the Octopi to the printer.

This threw me off somewhat, had to configure the print to start at 257C instead.

andrivet, 2017-12-27:
@Jannibal this is not a bug, this is expected. The current max temp is 260 degrees.

Jannibal, 2018-01-04:
@andrivet - I suspected as much. But I wanted to raise the issue anyway, in case it was something iffy going on. Good to know that it wasn't anything weird, after all.

Keep up the good work!

TheFlaash, 2018-01-12:
I am printing from s3d 4.0 and ich have to restart the printer about 3 times because of that error till the bed is over 47.
Printing via usb 
target is bed 55 
hotend 225
and like i written every time wehen the printer was cold

xyzleon, 2018-01-20:
I updated to the latest ADVi3 this week and I can't heat the bed or hot-end without getting the Thermal Runaway error. As soon as I hit pre-heat or set a temp using Octopi, I get this error and the unit hangs.
Even temps as low as 30 as a test fail instantly.

I thought it was my rig as I had also done a few tweaks, but flashing the original Wanhao firmware on and the printer heats up normally with no errors reported.

andrivet, 2018-01-24:
The original Wanhao firmware does **not** have temperature protection. The idea of temperature protection is that if a thermistor fails, it gives a low value to the firmware that will increase, increase, ... the temperature until your printer burns. So if ADVi3++ detects something unusual, it shuts down the printer for your safety.
In your case, one of your thermistors is apparently outputting some unusual value.
In the next release, the Thermal Runaway screen will give more details about what is wrong (reason and temperature).

jamesarm97, 2018-01-24:
Some firmware has a rise (C) over time, not just a temperature that is above a certain value (max temp). I had a problem on an unrelated firmware (Reprapfirmware) and I had to adjust the expected rise over time because my large heated bed was not heating fast enough. Could be something similar if your temperatures are showing ok (just rising too slowly or too fast)?

andrivet, 2018-01-24:
ADVi3++ does have rise over time. not just a temperature that is above a certain value. But since @xyzleon says that the screen appears immediately, it is not what triggers the error.

cyberhiker1965, 2018-01-26:
I am using version 2.1.0.

 Ran into this error on hour 17 of a 26-hour print.  Printing via USB with OctoPrint 1.3.6. Bed temp=65, Hot End Temp=215.  Have previously printed several ~10-12 hours prints.  It was in the middle of the night so the room temp would have dropped as well.  I captured a screen print which I will attach.

[image]

cyberhiker1965, 2018-01-26:
OK - same result on that reprint - this time earlier into the print.  The log shows...

`Recv: ok
Send: N584512 G92 E0*120
Recv: X:142.63 Y:113.62 Z:11.95 E:0.00 Count X:11553 Y:9203 Z:4786
Recv: ok
Send: N584513 M105*25
Recv: ok T:214.73 /215.00 B:60.56 /65.00 @:93 B@:127
Send: N584514 G1 X142.630 Y113.617 E0.0001 F4500*12
Recv: ok
Send: N584515 G1 X142.631 Y113.619 E0.0001*101
Recv: ok
...
Recv: ok
Send: N584597 G1 X144.139 Y114.490 F4800*114
Recv: ok
Send: N584598 G92 E0*122
Recv: Error:Thermal Runaway, system stopped! Heater_ID: bed
Changing monitoring state from 'Printing' to 'Error: Thermal Runaway, system stopped! Heater_ID: bed
'
Recv: Error:Printer halted. kill() called!`

And here's a screenshot:

[image]

Hope this helps a little.

andrivet, 2018-01-26:
The graphs you send clearly show that you have a temperature stability problems. The fluctuations are abnormal and this is why ADVi3++ shuts down the printer. This could be a defective thermistor or perhaps it has moved or it is not enough in contact with the bed. But this is not a problem in ADVi3++.

cyberhiker1965, 2018-01-26:
Any commands to enable PID tuning of the heated bed or is that disabled by default? I tried "M303 E-1 S60 C8" based on (http://marlinfw.org/docs/gcode/M303.html) but got an error "Recv: PID Autotune failed! Bad extruder number".

andrivet, 2018-01-26:
It is not enabled in the firmware (i.e. PIDTEMPBED is not defined). PID tuning of the bed for this particular printer seems to be unreliable.

xyzleon, 2018-01-26:
It was cold outside when I attempted to use the printer recently so the hot end read nearly zero when I started and the thermal run away error appeared. 

Switching instead to Wanhao v3 firmware, it heated  up from this same temperature and did a 7 hour print. Both hot end and bed temp were stable according to Octoprint for this duration so I'm confident there is not an issue with my printer.

Maybe advi3 saw the low starting temps as an error?

Cskull, 2018-01-27:
xyzleon where did you find this Wanhao v3 firmware

Cskull, 2018-01-27:
i too had a thermal runaway error twice once before the pid and once after it was on the preheat setting two   did not get that far into it  one time was from the machine control panel the other was from simply 3d machine control  because i had to re flash my machine to stock how ever when i did it some how corrupted the LCD screen it stays light up but black with a few thin bars that are blue

xyzleon, 2018-01-27:
Cskull - from here:  https://github.com/Silverquark/i3PlusPlus/tree/master/OriginalFirmwares

andrivet, 2018-01-27:
Bed min temp is 5°C. Any lower value will stop the printer.
@xyzleon @Cskull: Please do not use this issue to "chat". This is not the right place.

Cskull, 2018-01-27:
Dont worry about it that ill never use your software again or post on your
site

xyzleon, 2018-01-27:
I thought everything I had written was trying to be helpful and relevant to this issue but I'm happy to stop helping.

I will not respond to any further issues or posts - 
I'll stick with the Wanhao v3 firmware as it works. 
Thanks again.

gspitman, 2018-01-28:
I'm getting runaways before the bed finishes heating. Target temp 100c failure at 80c no other changes but firmware flash

gspitman, 2018-01-28:
I'm on your latest release

andrivet, 2018-01-28:
For the moment, I never see clear indication that there is a problem (a bug) with ADVi3++ "Thermal Runaway". It works as expected. However, it is also clear that some printers are behaving strangely with big temperature fluctuations (maybe a faulty thermistor, a the thermistor has moved, ...) and the shutdown of the printer is annoying for these people. I will see if there is a way to enable/disable "Thermal Runaway" programmatically (option of the LCD screen). I already see that it is not easy (this is driven by macros in the source code), but I will try.

cyberhiker1965, 2018-01-28:
I did some research on my heated bed yesterday. The thermistor is securely attached. When doing a preheat for an extended period the bed temp remained stable. However when the bed is moving during a print the reported temp fluctuates. Perhaps it generates enough air movement that it reports a little off? Or it could be some wire stresses? I've ordered a replacement and will report back once I get it installed.

jvskill, 2018-01-31:
I'm having similar issues with the Thermal Runaway Error stopping every print and PID Tuning. I can sort of get to the target temp of my prints but the second the extruder moves, the temperature reads one bad temp and shuts down the printer.

Maker Select Plus
ADVi3++ 2.1.0
Cura 3.2 Beta
Micro Swiss All Metal Hotend w/ slotted cooling block
DiiCooler

-Happens on both SD card, USB Prints and built-in PID Tuning
-Cura 3.2 Beta
-Target Temps: Extruder: 195C,  Bed: 60C
-Inconsistent, sometimes crashes from 125C-175C and sometimes gets to target temperature but movement causes a temporary temperature drop.

Here's a video of it happening in the latter case: https://youtu.be/VI7HO83B2PU

andrivet, 2018-01-31:
@jvskill Thank you a lot for the video. It helps me understanding what is going on. I do think that your printer has a hardware problem (heater thermistor under the washer is moving - perhaps because the screw is loose - or because the connector is not well plugged). The firmware is just reading the temperature. There is no processing of temperature at all. So a drop like that is abnormal.

However, I now see a pattern, not only for this printer, but for all printer based on a recent version of Marlin. Marlin does not tolerate such sudden drop of temperature. And I think it has. It may be possible to adjust existing parameters or I will have to modify Marlin itself. I will try.

jamesarm97, 2018-01-31:
I think Repetier did multiple reads and averaged them, I haven't looked at how Marlin does temperature reads and processing but if they act on a single read then noise would instantly make a fault like that.

jamesarm97, 2018-01-31:
I see Repetier also debounces the alarm tests so it takes multiple reads of a bad temperature. This would be a simpler fix assuming we are just seeing very short blips. 

            if(extruderTempErrors < 10)    // Ignore short temporary failures
                extruderTempErrors++;
            else {
                act->flags |= TEMPERATURE_CONTROLLER_FLAG_SENSDEFECT;

andrivet, 2018-01-31:
@jamesarm97 I do not want to reinvent the wheel and will try to use as much as possible existing Marlin code (otherwise, it will be more difficult to migrate to future Marlin versions). Marlin has more sophisticated tests (time-based, not counter-based) than what you suggest. Currently, ADVi3++ is configured to accept a decrease of 6 degrees during 1 minute. Apparently, it is not enough.

jvskill, 2018-01-31:
That's not what I see happening in the video or other crashes. These single readings trigger the error immediately. Some function might be out of order allowing a single bad temperature reading to bypass the 1 minute average. I will inspect my thermistor, but this may not fix the issue of electromagnetic noise that is inherent to the printer itself and the power source. The printer shares an outlet with 2 pcs and another laser printer, so some noise cannot be removed.

I will try to provide more details of what's triggering my error.

andrivet, 2018-01-31:
I am not convinced that it is electromagnetic noise in this precise case. A simple solution is to offer a way to deactivate temperature protection (i.e. Thermal Runaway errors). It looks like it is possible with few modifications of the Marlin code.
But I have to figure out how to make a switch button on the LCD screen (I know it is possible)

andrivet, 2018-02-01:
I found in the source code of Marlin exactly what we need: MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED. Currently, it is not used. I will.

jamesarm97, 2018-02-01:
Good find, so it works just like the sample repetier code I posted from what I saw. Glad it is something simple and already in place.

Rynstrum, 2018-02-03:
Hey there, Using 2.1 and attempting to reach 265 temp for abs inter layer adhesion. Seems to work on my other 2.1 machines. Anyways, even with the raised temperature ceiling it still seems to fail to get above 260. Curiously, a pid tune seems to oscillate near 260 but still never reaches 265 and doesn't trigger a thermal runaway issue during. This is an older machine so It may be the thermistor but its acting like its a software ceiling like previous software versions. Thanks for any insight anyone can share.
[image]

jvskill, 2018-02-08:
I have recently reassembled my hot end assembly and installed a new cooling fan for the heating block. Since then I have not had this error. I haven't had any readings that show a temperature spike. I'm not sure if the issue was resolved from loose connectors or loose thermistor. 

Let me know if there are any questions.

JackBradach, 2018-02-18:
I am also seeing the thermal runaway error, but it seems like it only happens when I use the on-LCD controls to load the extruder.  The temperature graph in Octoprint doesn't look like it went outside of the 200 degrees setpoint when it shut itself off.

I just did an overhaul of my printer, including replacing Marlin with ADVI3pp and a new heater block.  It has a silicone sock and a screw-in thermistor now, versus the kapton-wrapped cotton insulation.  Maybe it's losing too much heat at the sensor now?  I'll try out MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED and see if I can catch it occasionally reading a bad value.

lmattpatt, 2018-03-09:
I recently did a overhaul of my Maker Select Plus.  All metal hot end, Diii cooler, replaced thermistor in printbed and upgraded to latest ADV.  I’m getting thermal run away notification when going above 270.  I’m printing a special blend of PETG that requires higher temps than normal PETG.  I’ve ran PID tune several times at 270 and it heats up and finishes.  Not quite sure what’s going on here but there seem to be a lot of people having similar errors.  All of us can’t have bad thermistors right?

andrivet, 2018-03-09:
@lmattpatt Max temp is 275 degrees for the firmware. So you probably exceed this limit at some point. I do not believe it is your thermistor.

lmattpatt, 2018-03-09:
Thanks for the quick reply.  Is there any way to increase it to 280?  Here’s a screenshot from Octoprint.

Matt

JackBradach, 2018-03-11:
Check out Configuration.h, "HEATER_0_MAXTEMP."  Change from 275 to
something high enough that it can PID around it (maybe ~290-300?).  Rebuild
and reflash.

- Jack

andrivet, 2018-03-11:
I don’t think that the thermistor is rated for such high temperature. So you take the risk to burn you printer. Also, if you do not have a full-metal hotend, more than 245 will probably burn quickly you PTFE tube.

JackBradach, 2018-03-12:
Oh, yes.  Very important safety tip. :-)

Let me rephrase:

"Provided you are certain that all components in your system can handle it,
you can use HEATER_n_MAXTEMP to change the limit."

I think my PTFE tube started to get trashed at like 250ish, for what it's
worth.

- Jack

lmattpatt, 2018-03-12:
Thanks guys.  I’m running an all metal hot end so PTFE tube isn’t a problem.  I went ahead and changed the thermistor on the hot end and re-ran PID calibration.  The temp doesn’t seem to fluctuate near as bad and I can run up to around 268 C and not have max temp kick in.  I haven’t tried it any higher as I’m now getting excellent prints from the high temp matte black PETG I’m using.

Matt

blazze11, 2018-03-20:
Thermal runaway to agressive maybe.

Settings:
Maker Select Plus
ADVi3++ 2.1.0
Cura 3.2.1
Micro Swiss All Metal Hotend w/ slotted cooling block
CiiCooler mod

Filament "Silver PLA DasFilament"
Temps: Hotend: 215°C, Bed 55°C.
After the first layer my Partcooling fan goes from 0 to 100%. At this moment my temperature drops from 215°C to 200° and the heater is to slow at heating it back to 215°C in time. So a thermal runaway happens. Is it possible to tweak the time a bit?

cnsheets, 2018-03-28:
I just started experiencing this problem yesterday while in the middle of a 17 hour print. This was my fourth 17 hour print in 4 days as I had just successfully completed the previous prints. Up until now, I have not experienced any problems. 

Settings:
Maker Select Plus
ADVi3++ 2.1.0
Cura 3.2.1
Micro Swiss All Metal Hotend kit with cooling block
DiiCooler mod

Filament: PLA
Temps: Hotend: 210°C, Bed 60°C.

I've been using this same exact product for months with the same exact settings print after print without issue until last night.

First thing I checked was the bed thermistor. I did a pre-heat of both the bed and hotend to match my print settings. Disabled the motors and moved the bed around and watched the graph. I saw an occasional blip in the reading, but nothing major. Then just suddenly I got the error. Then I did the same for the hotend. Again, got the error. Went back to checking the bed thermistor, moved the wire a couple times and noticed it would suddenly jump from 60c to -15c then error. I pulled the bed thermistor and tested it to find the wire partly broken inside the insulation. Fixed that, tested good, re-installed.... but continue to get the same error. Short of buying new thermistors, I'm not sure what else could be the problem.

DylanFahey, 2018-04-04:
I get this error a few minutes after I start printing, it happens at nearly the same place on the print every time. I'm pretty sure this is due to my temps dropping rapidly due to an uninsulated hot end. 

Printer: Powerselect i3 V2 (Wanhao I3 Plus)
Firmware: 3.0.0-Beta 1

Temperatures
Bed: 60 C
Hot End: 220 C
Ambient: 20C (or perhaps slightly lower)

I'm using the stock cooler with an adapter which allows me to use a radial fan. 

Print Source: SD Card

The temperature will drop to about 200 degrees C during printing when the cooling fan is running at 100%. Turning down the fan will result in progressively warmer temps. 

I was having issues with jamming and I needed to take the printer apart a few times, so I removed the insulation. Removing the insulation actually fixed my issue with jamming, so I suspect heat creep may have been the issue. 

I suspect the temps dropping so quickly has to do with where the stock cooling duct is aimed and the increased airflow from the radial fan. I have a silicone sleeve on the way which may solve the problem.

gspitman, 2018-04-09:
has 3.0 been released?

andrivet, 2018-04-09:
3.0.0 Beta 2 has been released