andrivet/ADVi3pp #45: Fan speed on the print temperature log shows zero.

rippledabs, 2017-12-03:
Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: Fan speed shows "0" when the g-code turns on the fan. 
- Expected behaviour: Should show the fan speed. Also, when manually adjusted, it's missing the % symbol like before. 
- Actual behaviour: Shows zero.
- Steps to reproduce:
  - Print something

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

andrivet, 2017-12-07:
I am not able to reproduce what you see. In my test, the update is quick (1/2 second). Can you give more details:

- Exact version of ADVi3++?
- Are you printing from the SD card or using USB?
- In the case of USB, with which software (name and version)?
- How do you change the fan speed? In the printing software? In a plug-in?

andrivet, 2017-12-09:
Close because no feedback. Be free to comment if you want to reopen issue

closed 2017-12-09 by andrivet