andrivet/ADVi3pp #49: [FR] Start a development branch

andyfarthing, 2018-01-03:
Would it be possible to start a development branch going?

There's a few of the more straight-forward things from the FRs that people can implement and make pull requests for, but at the moment we don't really know what's already been sorted privately and so what's worth spending time on. 

For example, I've implemented the BLTouch code from @weidercs' fork, but I've made a few tweaks that speeds up the probing by a fair bit, and which should improve reliability of the readings. I'd make a PR for this, but it looks as though you've already implemented BLTouch privately for testing? My changes are also based on 1.1.6, which is the latest version we have access to on this branch. It looks as though some changes to levelling have been made thru 1.1.8, so they might not even be relevant anymore. 

The main aim is to prevent people from putting in a lot of work, only to find out that it's already been taken care of elsewhere (such as @voretaq7's PR from the other day). It also makes sure that people who *are* working on things are working with the latest sources, so we're not adding things that are no longer relevant or required.

Either way, thanks for the work you've put into this so far - it's definitely one of the more promising firmwares that have been worked on.

sniperxnl, 2018-01-03:
Do you have a workout for the bltoutch?

voretaq7, 2018-01-05:
I'd be all for more feature branches in the repo (even if they're unstable) or committing to the head with tags for stable releases if that's your preference - lets us test with the latest code & avoid duplication of effort.

andrivet, 2018-04-01:
Development branch is called advi3++dev.

closed 2018-04-01 by andrivet