andrivet/ADVi3pp #52: [FR] Support of LJ18A3-8-Z/BX 8mm Approach Sensor

andrivet, 2018-01-20:
### Feature Request

Support of LJ18A3-8-Z/BX 8mm Approach Sensor:
https://www.aliexpress.com/item/Promotion-LJ18A3-8-Z-BX-8mm-Approach-Sensor-NPN-NO-Switch-DC-6-36V/32623086326.html

alexten9, 2018-03-12:
Hey by any chance did you add the feature? And if so is there a good how to install it? Thank You for your work!

andrivet, 2018-03-13:
My priority for version 3.0 is BLTouch support. Support of other sensors (I bought 2 of them) are for a future release (probably 3.1).

andrivet, 2020-01-19:
https://community.advi3pp.com/t/can-i-just-overwrite-firmware-of-i3plus-with-inductive-sensor-with-advi3-with-no-bltouch/762/10

andrivet, 2022-07-15:
It is far better to use a BLtouch already supported by ADVi3++. So I will not implement this.

closed 2022-07-15 by andrivet