andrivet/ADVi3pp #55: Reproducible white screen crash

timangus, 2018-02-06:
- Description: Crash to white screen with no ability to control the printer.
- Expected behaviour: Return to the main menu from the print temp monitor screen
- Actual behaviour: White screen, no control of printer.
- Steps to reproduce:
  - With **no SD card inserted**, power on the printer
  - Press Print
  - Press Controls
  - Press back

andrivet, 2018-02-06:
I am not able to reproduce the behavior: if I power on the printer and there is no SD card inserted, when I press Print, I have the Temperature screen without controls.

The only way I found to reproduce something close is to insert the SD card start a Print, remove the SD card, etc.

timangus, 2018-02-07:
Hmm, that's odd. Something weird about my printer/setup/install? https://youtu.be/OvLxc6Qw6ZI

andrivet, 2018-02-07:
I did not understand that you press the two icons so quicly. With the video, it is clear. It is a bug in ADVi3++

timangus, 2018-02-07:
Ah cool. Sorry, probably should have made that clear.

andrivet, 2018-04-08:
Fixed in Beta 2

commit 62b124aa (2018-04-08): Fix #55: Reproducible screen stucked. Better fix for #54.
closed 2018-04-08 by andrivet