andrivet/ADVi3pp #57: USB printing graph display does not show pause, stop, etc. printing via usb (Repetier host/server)

TTVert, 2018-02-12:
**Description: **-
 When printing from repetier host (pi connected via USB) print graph does not show 
**Expected behaviour:**
Expect to see this: https://andrivet.github.io/ADVi3pp-User-Manual/assets/044_print.png
**Actual behaviour:**
See this: https://www.dropbox.com/s/wbsgbdsrrdrdzni/2018-02-12%2012.16.21.jpg?dl=0
**Steps to reproduce:**
 Print any document from repetier host to repeteir server

jamesarm97, 2018-02-12:
I think you are limited as to what you can do if printing from USB. How would you pause or stop on the printer side when the data is just coming in as a stream of gcode? That is also why some commands like the gcode command to pause only works when printing from SD. The firmware is in control of reading the gcode from a file on the SD and can just stop processing (pause) and wait for resume command where if the data is coming in over the USB, how would you tell the Repetier Host to stop sending the data? My thoughts though.

TTVert, 2018-02-12:
After looking a bit it seems this may have been pulled in 2.x.  Their docs state that while printing via USB that is the screen you should see which is why I was confused when I didn't see it.

https://github.com/andrivet/ADVi3pp-Marlin/issues/31

jamesarm97, 2018-02-12:
The only thing I can think of to enable pausing would be if the host (Repetier Host) is set to Wait for OK before continuing then if you pause on the LCD, the firmware will just clear the buffer and not send any more OK's back then if you press Resume it will send one OK which should start the USB data again. Not sure if this would be worth looking into because if you have a host right there, just press pause on the host. Wait for OK might not be something that every host implements.

TTVert, 2018-02-12:
Possibly not.  In my situation I have my printer connected to my pi via usb which runs repetier server so the printer and the pc I use to initiate print via repetier host may be 100s of feet away.  It'd be nice to pause or stop a job that is having some issue from the control panel.  I am trying to recall if my i3 (non plus) allowed for this.  

Dave

jamesarm97, 2018-02-12:
It doesn't. I am printing using Repetier Server right now on an I3 (not plus, running Repetier Firmware)  and you have no way to stop locally except pressing a reset button (if you drilled a hole to gain access to it).

andrivet, 2018-02-12:
This works as expected. Such controls are only available for SD printing. I however see that my User Manual is wrong. I will fix it.

closed 2018-02-12 by andrivet