andrivet/ADVi3pp #59: [FR] Manual Mesh Bed Leveling

bobbydamaker, 2018-02-18:
------------------------------------------------------------
### Feature Request

I saw that you have added milestone for automated bed leveling, but I was hoping to add a request to also allow access to the manual and mesh bed leveling (G29 Command) that is available in Marlin.

Mooncabbage, 2018-02-27:
This would be a useful additional feature under the Tuning menu. I haven't tried it, but one of the reasons I installed this firmware was to do so.

What would the manual mesh leveling look like? I'm not sure what's involved, but knowing what sort of features the software would require is probably a good starting point.

fedemartino, 2019-08-29:
Is this still a possible enhancement? This would be really useful!

Darkhand81, 2020-08-08:
As a Patreon backer, just wanted to comment that I'm really excited to see that you added this to the 5.0 milestone!  I've been excited for manual mesh leveling for a long time. Please take this as a huge support behind the feature!

andrivet, 2025-02-12:
Implemented in 6.0.0

closed 2025-02-12 by andrivet