andrivet/ADVi3pp #62: Inconsistent Extrusion Tuning

Mooncabbage, 2018-02-27:
### Bug Report
- Description: Extruder testing on PLA <210 degrees gives inconsistent results.
- Expected behaviour: Repeatable results at 200 degrees, 190 etc.
- Actual behaviour: The exact same E-steps value will sometimes extrude 10-20mm over, sometimes 10-20mm under, impossible to get closer than 5mm despite numerous adjustments. Testing at 210 seems to produce a consistent result of approx. 100 E-Steps.
- Steps to reproduce:
  - Run Extruder tuning with no changes, PLA.

Not sure what settings the test uses for feedrate, however in my own testing, the recommended 100mm/m by most guides does not work with the stock extruder setup. When I configured it over USB with stock firmware and RepetierHost, I got my most consistent results with: 
```
M109 S210
G92 E0
G1 E100 F50
```
Lower Temps may also work with this feedrate, I did try them but I don't remember the results.

andrivet, 2018-02-27:
The firmware does the following:

```
M83
G92 E0
G1 E1 F100
G1 E1 F100
...
G82
G92 E0
```

(I think that your `E100` was a typo).
So yes, I am using 100 mm/m. I will make tests to see if I can reproduce the same problem and if a lower speed helps.

jamesarm97, 2018-02-27:
I think when we did the script for the I3 (v1/v2) we ended up posting that F50 worked best for most. Slow and steady wins the race. 

You can run the nozzle at a hotter temperature. You want the filament to flow as freely as possible for this test, you are not really trying to repeat it at the temperature of the filament spec. Just needs to be run once without skipping.

Mooncabbage, 2018-02-28:
It's not a long test, slowing the flow rate drastically reduces skipped steps and other issues. E100 is not a typo, as you I think worked out. I'm not a G-code expert, but I would imagine that G1 E100 F50 is a bit more efficient than G1 E1 F50 * 100 times, and it's a lot easier to bug test!

andrivet, 2018-02-28:
@Mooncabbage I realize after that it was not a typo and your command was right. I am doing several small commands to enable canceling the process. In my computation, I take into account the actual amount extruded. But maybe it is not the right way to do this. So I will perform tests and I plan to fix this problem for version 3.0

andrivet, 2018-04-08:
Temporary fix in Beta 2. Canceling will be implemented in a later release.

commit 2f321847 (2018-04-08): Replace Extruder calibration #2 by the waiting page. Fix #62: Inconsistent Extrusion Tuning
closed 2018-04-08 by andrivet