andrivet/ADVi3pp #63: [FR] Add the LCD model in the About screen

andrivet, 2018-02-27:
Since today, there at least two different models of the LCD screen for this printer, detect and report the model number (can help in case of a bug specific to a model).

SureshotM6, 2018-03-02:
R0 on my DMT48270M043_02WT contains 0x00.  Unfortunately, it is very likely that is not valid since that overlaps with a different LCD model in the manual.  Could you see what your _05W returns, as it is not listed in the manual either?  I added this code at the end of get_lcd_firmware_version(), which I can submit a pull request for if you get something other than 0x00:

```c++
String PrinterImpl::get_lcd_firmware_version()
{
    ...

    frame = ReadRegisterDataRequest(Register::R0, 1);
    frame.send();

    if(!response.receive(frame))
    {
        Log::error() << F("Receiving Frame (R0)") << Log::endl();
        return lcd_version;
    }

    response >> version;
    lcd_version << " " << hex_byte(version.byte);

    return lcd_version;
}
```

The code definitely works, as I am able to read other registers such as R1 as the correct values.

andrivet, 2018-03-02:
Already made the test. Returns also 0

andrivet, 2018-03-02:
In fact, R0 does not give the model number. It is the "screen mode" and is probably always zero. These screens are not DGUS but Mini DGUS and there are some differences.

closed 2018-03-02 by andrivet