andrivet/ADVi3pp #65: Boot Loop

agentnoise, 2018-03-05:
### Bug Report
- Description: When flashing the firmware marlin gets into a boot loop.  About every 5 seconds marlin issues a start command.  I am not able to send commands to the printer but I can flash it and I read the serial output. I do not have a problem with other firwares such as the original Wanhao firmware or Klipper. My LCD is broken so I cant reconnect it, not sure if that could cause an issue. I've tried clearing the eeprom and disconnecting all the hardware but it still does it. 
- Expected behaviour: --- Actual marlin boot
- Actual behaviour: --- Boot looping
- Steps to reproduce:
  - Flash the most resent firmware on the mainboard
  - Check the serial monitor to see if it keeps restarting

andrivet, 2018-03-05:
This firmware can't work without the LCD part.

closed 2018-03-05 by andrivet

andrivet, 2018-03-05:
For info, you can find a compatible LCD panel (DMT48270M043_05WT) for $ 25 on AliExpress (for example).

brandonjank, 2018-03-05:
Instead of dismissing entirely, it would be more user friendly to add a check for the screen and then send a terminal message letting the user know the screen must be connected. Or possibly go into a no-display mode for the few users may want to use your firmware without the screen features. :D

jamesarm97, 2018-03-05:
Just curious, isnt using this firmware without the Screen going to be just like using Marlin vanilla? If you know you are going to not be using the screen then use Marlin. The only case I can see if if you have the thing opened and the screen removed and happen to be trying something.

andrivet, 2018-03-05:
@jamesarm97 On boot, the firmware retrieves information from the LCD panel and instructs it to start the boot animation. So if there is no panel, the communication will fail (and reboot the firmware - Marlin does not like such unexpected things). 
@brandonjank It is not obvious to handle this very specific case. I can spend time handling this issue but I prefer to spend time fixing other issues (that impact all users). 
I have closed this issue because I will not implement a fix and I use the list of open issues to know what I have to do.