andrivet/ADVi3pp #66: [FR] Moving E steps without heating up extruder.

alexten9, 2018-03-13:
Sorry was not sure if this is already possible. If so please let me know how.

Im trying to calibrate E steps and I have the motor removed from the nozzel. The motor wont move unless I preheat the nozzel to c180. Is there a way to calibrate it by bypassing the nozzel?

jamesarm97, 2018-03-13:
Sending the following (via USB serial, repetier host, octoprint or other terminal means) may work. It is a standard marlin gcode. P1 disables the check.

M302 P1      ; disable cold extrusion checking

andrivet, 2018-03-13:
@alexten9 This is not a feature request. Please do not use issues for support questions.

closed 2018-03-13 by andrivet

alexten9, 2018-03-13:
It woupd be nice if it coupd be done on the printer it self. Call it cold e step calibration.