andrivet/ADVi3pp #67: Temperature under-run when lowering target

csutcliff, 2018-03-31:
### Bug Report
- Description: When you lower the target temperature mid-way through a print (automatically or manually) the actual temperature drops a lot lower than the new target before climbing back up.
- Expected behaviour: Temperature drop shouldn't over-shoot so far. If this is not possible printer should wait until the temperature stabilises before continuing with the next layer.
- Actual behaviour: If I print first layer at 200degC before lowering to 190degC the temperature drops to around 178degC before climbing back to 190degC
- Steps to reproduce:
Start a print at a higher temperature for the first layer and see what happens when the temperature changes.

If it helps I'm using the complete Micro Swiss extruder set up and I have done PID tuning. Temperatures are very stable apart from this behaviour.

andrivet, 2018-05-21:
Do you still see this behavior? I have tried many times from different temperature. What I see is a decrease of temperature (4 to 5 deg under the target) and then it stabilize at the target temp. I never see what you observe. Do you have the fan on?

andrivet, 2018-05-24:
No news so close this issue for the moment.

closed 2018-05-24 by andrivet