andrivet/ADVi3pp #72: Error when compiling source within arduino

xKoldx, 2018-04-03:
Once I've downloaded the source code and renamed marlin.cpp to marlin.ino, I tried to compile and got an error message within Arduino 1.8.5. ive tried the suggested conversion of the encoding with no luck. 

`Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\advi3pp.cpp: In member function 'void advi3pp::internals::Printer_::sensor_grid_show()':

advi3pp.cpp:1959: error: 'GRID_MAX_POINTS_X' was not declared in this scope

     for(auto x = 0; x < GRID_MAX_POINTS_X; x++)

                         ^

advi3pp.cpp:1960: error: 'GRID_MAX_POINTS_Y' was not declared in this scope

         for(auto y = 0; y < GRID_MAX_POINTS_Y; y++)

                             ^

advi3pp.cpp:1961: error: 'z_values' was not declared in this scope

             frame << Uint16(static_cast<int16_t>(z_values[x][y] * 1000));

                                                  ^

exit status 1
'GRID_MAX_POINTS_X' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

biuklija, 2018-04-03:
Have you defined/enabled or disabled BLTouch support? The BLTOUCH define at the beginning of configuration.h is only for CLION, so you have to put 
#define ADVi3PP_BLTOUCH 
somewhere else if you plan on building with it.

andrivet, 2018-04-03:
You are taking Beta sources, so everything is not finalized. The same sources can be compiled with or without BLTouch, so ADVi3PP_BLTOUCH is not defined in the sources. It has to be defined when compiling.

xKoldx, 2018-04-03:
Yupp, Your right. sorry, I didn't realize what was missing. I like where you're going with this though! 

It's working now!

ill mess around with it and see if I find any bugs.

closed 2018-04-03 by xKoldx