andrivet/ADVi3pp #73: FR: Enable LIN_ADVANCE?

vertigo235, 2018-04-03:
I would like to experiment with the Linear Advance feature within marlin, I see that your firmware build has it disabled. Reading the documentation it looks like you can enable it, but set the default K value to 0, that should enable us to experiment with the settings. 

#define LIN_ADVANCE

#if ENABLED(LIN_ADVANCE)
#define LIN_ADVANCE_K 0

I'm just reading into this, and I'm not really sure how to compile my own firmware, if you do not think enabling this in your releases, can you tell me how you are compiling your source and I will look to compile my own firmware?  Thanks for your awesome work.

vertigo235, 2018-04-08:
I think so , I looked at the advanced config .h file that the dev used for the firmware and the linear advanced function is commented out. So that tells me it should be disabled. 

Maybe the dev can give us some confirmation.

andrivet, 2018-04-09:
@f0lie @vertigo235 Linear Advance is not enabled and will not be in the near future. At a minimum, not before the release of Marlin 1.1.9.

vertigo235, 2018-04-09:
Sounds good I saw that the 1.1.9 version it has changed and should be better anyhow.

andrivet, 2018-08-18:
My current idea is not to just enable LIN_ADVANCE and some random value for K, but to include a wizard to determine K. It may look like this:

[image]

When you press continue, it will generate the appropriate G-Code (similar to http://marlinfw.org/tools/lin_advance/k-factor.html) and start a print. You can then determine the best K and enter it

andrivet, 2019-05-26:
Implemented in version 4.0 but without the Tuning screen.

closed 2019-05-26 by andrivet