andrivet/ADVi3pp #77: ### Bug Report G28 Home

Titus142, 2018-04-05:
### Bug Report
- Description: Z homes on center of build place instead of relative position
- Expected behaviour: When G2 called at start of Gcode I expect to home Z at X and Y home, or if pressed on LCD to home Z in relative XY position
- Actual behaviour: XY moves to center of build plate them homes Z

Thanks for the AWESOME firmware, I really appreciate the work that goes into this!

andrivet, 2018-04-06:
This works as expected. It is not a bug. However, as it confuses people, I will see if I change it back to the previous position.

andrivet, 2018-04-07:
Home position is back to (0,0,0). Implemented in Beta 2

closed 2018-04-07 by andrivet

john-f-chamberlain, 2018-04-07:
Is it possible to have an option to use the BLTouch as the Z limit rather than the limit switch? I may be incorrect, but wouldn't that allow for the bed to be in a fixed position (no springs , no adjustment thumb screws) and guarantee that the head won't crash into the bed?

andrivet, 2018-04-07:
wherever xy is currently

Awesome. If i hit home z will it go to xy 00 or will it home wherever xy is
currently?