andrivet/ADVi3pp #79: Support for filament out / jam /error handling

Martin-Dolan, 2018-04-06:
It would be great to have support for a filament sensor that would pause the print when an error condition occurs.

Error could be simple filament out sensor:
 (many diy options or $3 from AliExpress
 http://s.aliexpress.com/bmYJzmyY )

Or something that detects filament stop / tangle errors:
Again, a few diy options or this one https://www.toybuilderlabs.com/products/tunell-3d-printer-filament-monitor

The function would need to pause print and lift print head up a predefined distance from print so it doesn't melt print.  Then lower down and continue when error corrected. (via continue /cancel button on screen?)

andrivet, 2019-02-16:
Will not be part of version 4.0. Implementation is not ready and I do not want to postpone version 4.0 further.

Martin-Dolan, 2019-06-09:
I know this is not implemented yet, however do you have any suggested sensor types, or GPIO pins to use etc?
That way I can do the hardware part this end so it's ready for when software supports it.
No rush, you're doing amazing things. I'm just about to install the release version of 4.0 tonight. very excited :)

andrivet, 2019-06-09:
@Martin-Dolan For support questions, please post on [ADVi3++ Community](https://community.advi3pp.com/c/get-help), not here. Do not take offense, but I will delete your comment (and mine) since I try to keep these issues as clean as possible.

andrivet, 2020-07-30:
Implemented in upcoming version 5.0

jeoriol, 2021-10-21:
All hands on this one.

andrivet, 2021-10-21:
This is not the right place to vote for features. Votes are exclusively on ADVi3++ Community: https://community.advi3pp.com/t/vote-for-advi3-version-5-x-new-features/.
Issues here are only for technical considerations.

andrivet, 2022-03-21:
Implemented in version 5.1

closed 2022-03-21 by andrivet