andrivet/ADVi3pp #80: Pin Outs for BL touch

Bigfellas, 2018-04-06:
Not an issue. Just wondering if you could post a photo of the pin outs for the BLtouch 

Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

andrivet, 2018-04-06:
Already published for Patreon sponsors. Other users will have to wait a little. I am closing this since it is not an issue or a feature request.

closed 2018-04-06 by andrivet

Bigfellas, 2018-04-06:
Great Please send me a link to your Patreon

andrivet, 2018-04-06:
www.patreon.com/andrivet