andrivet/ADVi3pp #81: [FR] Multiple PID tune memory, or auto PID tune.

Mooncabbage, 2018-04-07:
Would it be possible to add a feature whereby the firmware stores multiple PID profiles for different hot end temps, or they are generated on an as-needed basis? I find myself printing at different temps, even with the same filament, depending on what I want to achieve, or as I tweak my settings. Having the wrong PID tune seems to cause some fairly awful artifacts. It would be nice if the firmware could be a bit smarter about how to achieve/maintain a given temperature, rather than have to run the PID tune every time I want to change the hot end print temp.

lightseeker34, 2018-05-30:
This would be good feature to have. My new printers have this feature. It's becoming standard. During the pause, a Z+ movement would help keep the print clean.

andrivet, 2018-07-04:
it may look like this:

[image]

[image]

andrivet, 2018-11-04:
Also related to #101 and #39

andrivet, 2018-12-10:
Implemented in Alpha 1

closed 2018-12-10 by andrivet