andrivet/ADVi3pp #86: Cant change thermistor type

alokbhargava52, 2018-04-09:
Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: --- 

Can't change Thermistor type to 5 as needed to implement E3D extruder
- Expected behaviour: ---

It should be possible to easily change

- Actual behaviour: ---
- Steps to reproduce:
  - Open the configuration.h file
  - Change thermistor type to 5
  - Recompile
  - You see an error saying thermistortable_5.h missing 
  

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

andrivet, 2018-04-09:
There is no bug here. If you need to change the thermistor type, get the thermistortable_5.h from Marlin.

closed 2018-04-09 by andrivet

alokbhargava52, 2018-04-09:
Thanks. 

I had no idea that I had to add missing files from the original marlin 1.1.8. I did it and able to compile successfully. 

One question: I installed your hex files and also flashed lcd and it got installed successfully. On testing I found extruder motor doesn’t turn. I had done boden configuration with e3d heat block. 

Motor is ok. I think motor driver got burned. What must have caused it? I’m very careful person and go step by step. Ribbon cable is ok. 

Thanks,
Alok

Sent from my iPhone 8 plus, iOS 11.3

alokbhargava52, 2018-04-09:
I have ordered the new 5.4 mother board but is there a way to repair the old one? Drivers are fixed and not replaceable. 

Alok

Sent from my iPhone 8 plus, iOS 11.3

andrivet, 2018-04-09:
@alokbhargava52 1) No idea. 2) 5.4 motherboard? Are you sure it will be compatible with your printer. Do you have a link to this motherboard (where you ordered)?

alokbhargava52, 2018-04-09:
https://www.reprap.me/wanhao-main-board-to-duplicator-i3-plus-v5-4.html

Sent from my iPhone 8 plus, iOS 11.3

andrivet, 2018-04-09:
I think that this is for Mark II (the new i3 Plus). And it is NOT compatible this i3 Plus version 1. Look at the connector for the flat cable: there are 2x10 pins instead of 2x8. You need version 5.1

alokbhargava52, 2018-04-09:
Oh. Thanks for alerting me. 

Alok

Sent from my iPhone 8 plus, iOS 11.3

andrivet, 2018-04-09:
You probably need this: https://www.reprap.me/wanhao-main-board-to-duplicator-i3-plus.html

alokbhargava52, 2018-04-09:
Thanks, you are very helpful.

Thanks,

Alok Bhargava
925-963-2565 (M)