andrivet/ADVi3pp #87: Printer Reset when using PID

xKoldx, 2018-04-09:
Beta V2, Just flashed. M502 and M500. 

When using The option under Settings>PID , Printer Resets. 

Serial Monitor Shows 

> [...]

Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo:Marlin 1.1.8
Recv: 
Recv: echo: Last Updated: 2018-02-16 12:00 | Author: (andrivet)
Recv: echo:Compiled: Apr  9 2018
Recv: echo: Free Memory: 1912  PlannerBufferBytes: 1232
Recv: echo:V48 stored settings retrieved (744 bytes; crc 22624)
Recv: echo:  G21    ; Units in mm
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X81.00 Y81.00 Z400.50 E94.30
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X450.00 Y450.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E1000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R800.00 T800.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E1.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P33.41 I1.47 D189.27
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.88
Recv: echo:SD init fail
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N0 M110 N0*125
Recv: ok
[...]

andrivet, 2018-04-09:
I am not able to reproduce this behavior.

xKoldx, 2018-04-09:
Ill reflash and see what happens.

xKoldx, 2018-04-16:
Reflashed both the LCD and the Board and still have the same results. 
Ive attached the only two files ive changed within marlin. 
[config.zip](https://github.com/andrivet/ADVi3pp-Marlin/files/1913553/config.zip)

xKoldx, 2018-04-16:
Figured it out. 

I defined the "#define ADVi3PP_BLTOUCH" Twice in the code, didn't realize you added this between builds.

xKoldx, 2018-04-23:
[config.zip](https://github.com/andrivet/ADVi3pp-Marlin/files/1936909/config.zip)

andrivet, 2018-04-25:
You configuration is very different from mine. This is not an issue of ADVi3++ but of your modifications so please do not open an issue here.

closed 2018-04-25 by andrivet

xKoldx, 2018-04-25:
I've uploaded the configuration using your default settings via arduino and
I still get the same results.

Keeping the pin settings and everything.

That's why I assumed it was a bug.
I'll double check but I figured I'd let you know.

andrivet, 2018-05-05:
Still not able to reproduce.

xKoldx, 2018-05-05:
When I flashed beta 3.0 issue disappeared. Didn't reply because issue was
closed.

xKoldx, 2018-05-05:
Thank you for your help!