andrivet/ADVi3pp #88: Printer Reset when navigating Menus

xKoldx, 2018-04-09:
When Homing the printer Via Controls>Move>Home All

If the printer is Homing when trying to access the menu for any other commands the screen will lock up and the printer will reset. 

I have modified the code to use "Z_Safe_Homing" as I am not using the Z_MIN_ENDSTOP. 
Just the BLTouch
I don't know if this is related. 

This also happens if you try to navigate too quickly. 

I'm not sure if this is a bug, or I just need to slow down. I'm just excited. 

This is trying to access the load and unload page while homing
> 
[...]
Recv: echo:enqueueing "G28"
[...]
Recv: echo:busy: processing
Recv: echo:enqueueing "OUNT:1 UUID:44b2f5d6-e7e4-47b"
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Recv: echo:Marlin 1.1.8
Recv: 
Recv: echo: Last Updated: 2018-02-16 12:00 | Author: (andrivet)
Recv: echo:Compiled: Apr  9 2018
Recv: echo: Free Memory: 1912  PlannerBufferBytes: 1232
Recv: echo:V48 stored settings retrieved (744 bytes; crc 22624)
Recv: echo:  G21    ; Units in mm
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X81.00 Y81.00 Z400.50 E94.30
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X450.00 Y450.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E1000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R800.00 T800.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E1.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P33.41 I1.47 D189.27
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.88
Recv: echo:SD init fail

------------------------------------------------------------------

This is when trying to access the tuning menu during Home All

> 
Recv: echo:enqueueing "G28"
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: �\x1f�\x0f�\x1f�\x01�_�O`�q�����/�\x0f��\x0e�_�O(�9�J�[�\x0f�M\x0f����\x08�\x0e��S�#\x19�\x0e��������\x0b���\x0b�+��\x08�\x1f�\x0f�\x0f�\x0f�\x11$\x0b�\x0f�����������\x0f�\x1f�/�?�O�_�o�\x7f�����������������
Recv: l\x17@�m\x17P�n\x17���.�,\x1c��.�,�,{\x01�\x01a�{���������C�Q�b�q���2�,���\�����\x0f�\x1f������\x01����\x01�\x19�\x0e��g\x04�j�{�����\x0e�\x1f�(�9�`\x1bq\x0b�\x0b�\x0b(�3�@�P�\x0f�i\x15���������\x01�\x01b\x0fs\x1f�\x1f�\x1f����\x1f�\x0f�\x08������\x01 �1�!01\x05i��\x01��������\x0e��gj�{�������\x01�������\x08��\x0f�\x1f�\x0f�\x1f�\x01�_�O@�Q�b�s��\x01�^�O@�Q�b�s����\x0b���\x0b�����������\x0b���\x0b���\x0b���\x0b�������\x0b���\x0b\x08�����/����\x01�\x0f�\x1f�\x0f�\x1f�\x01�_�O`�q�����/�\x0f��\x0e�_�O(�9�J�[�\x0f�M\x0f����\x08�\x0e��S�#\x19�\x0e��������\x0b���\x0b�+��\x08�\x1f�\x0f�\x0f�\x0f�\x11$\x0b�\x0f�����������\x0f�\x1f�/�?�O�_�o�\x7f�����������������start
Recv: echo:Marlin 1.1.8
Recv: 
Recv: echo: Last Updated: 2018-02-16 12:00 | Author: (andrivet)
Recv: echo:Compiled: Apr  9 2018
Recv: echo: Free Memory: 1912  PlannerBufferBytes: 1232
Recv: echo:V48 stored settings retrieved (744 bytes; crc 22624)
Recv: echo:  G21    ; Units in mm
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X81.00 Y81.00 Z400.50 E94.30
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X450.00 Y450.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E1000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P800.00 R800.00 T800.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X8.00 Y8.00 Z0.40 E1.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:PID settings:
Recv: echo:  M301 P33.41 I1.47 D189.27
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-1.88
Recv: echo:SD init fail

andrivet, 2018-04-09:
This is not related to your modifications. It is a bug

andrivet, 2018-04-22:
I do not remember why I confirm the bug. I had made many tests and I am not able to reproduce it with Beta 3.