andrivet/ADVi3pp #95: [FR] Revert SD card file sorting to recent first

biuklija, 2018-04-12:
-----------------------------------------------------------
### Feature Request
Files on the SD card, after folders, are sorted alphabetically. The old sorting method (newest files first) was much more practical because it enabled easier access to the last sliced file (which is most of the times the one you want to print)

------------------------------------------------------------

DynaMight1124, 2018-05-03:
Just to add a little bit onto this. It is a little annoying, I've recently upgraded from v2.1 to 3.0 beta and noticed my newly added files were not at the top as they've always been. Strangely enough they're right at the bottom but in then in date order with the newer ones since I upgraded at the 'top' of the list. A fix for this would be awesome. You can see theres been massive improvements between betas so thank you for that.

supergregg, 2019-06-26:
Yes i would like to ad to this. Maybe even a feature for selecting sort order? Like a button on the screen to cycle, Alphabeticly, Date and what ever more and a Default.

andrivet, 2019-06-30:
@supergregg No, this will not happened. It takes too much memory.

galfisk, 2022-03-10:
Then would it be possible to simply invert the order of file list from how it's currently displayed? That would in most circumstances put the newest files on top. Since people usually want to print the latest file added to the card, this'd be a nice usability improvement. The original Wanhao firmware does something like this.

andrivet, 2022-03-21:
I just checked and for a long time, ADVi3++ uses SDCARD_RATHERRECENTFIRST. As Marlin's configuration says:

> Reverse SD sort to show "more recent" files first, according to the card's FAT. Since the FAT gets out of order with usage, SDCARD_SORT_ALPHA is recommended.

So most of the time, files are listed in reverse order (latest first). But this is dependent of what your are doing with your SD card. If you are often removing files from it, the files will be in any order. The solution: reformat the SD card and put back the files your want.

closed 2022-03-21 by andrivet