andrivet/ADVi3pp #98: [FR] Support for 3 point leveling in Manual leveling

tobashadow, 2018-04-15:
Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

Needs to only have right side front and rear and left side center and the center bed position to do it.
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

andrivet, 2018-04-15:
Please give a minimum of information (such as reference, etc.)

tobashadow, 2018-04-15:
This is a common mod to the I3 Plus utilizing this plate - 

https://cdn.shopify.com/s/files/1/0002/9445/3292/products/PYP4-3_110x110@2x.jpg?v=1519236101

You can see the hole position on the left hand side where the 3 point level for that side goes.

andrivet, 2018-04-15:
Sorry, still not getting it. How it works?

tobashadow, 2018-04-15:
Ok here's a topic and good read on why 4 point leveling is wrong and why 3 point is better ( I fully agree after moving to it) -

https://drmrehorst.blogspot.com/2017/07/3-point-print-bed-leveling-vs-4-point.html

On a I3 Plus or V2 for that matter all you do is purchase that upgraded Y carriage which is a popular upgrade anyways and rotate the stock heated bed CCW to the left 90 degrees and the third hole that is already drilled in your bed lines up. Remove the two bolts and springs on the left side and move one to the third hole and level as given in that article and as long as you haven't bent your board alreadydoing 4 point leveling, it will stay flat from then on.

tobashadow, 2018-04-17:
I just realized that a way around this is to change the manual leveling spots on the screen from 5  like this currently

X      X
    X 
X      X

To

X  X  X
X  X  X
X  X  X

This would not only support 3 point leveling with the 3rd point in any position but also add extra check points for leveling even in 4 point leveling

andrivet, 2018-04-18:
@tobashadow This is probably the way to go.

andrivet, 2018-07-04:
It may look like this:

[image]

andrivet, 2019-05-26:
9 points implemented in version 4.0

closed 2019-05-26 by andrivet