andrivet/ADVi3pp #99: [FR] No more relying on the Z-endstop

andrivet, 2018-04-15:
Today, ADVi3++ is still using the physical Z-endstop for Z-height measuring and because I was not confortable with the idea to remove the stop (I see it as a safeguard). However, I get some feedbacks suggesting this endstop is not necessary, or may even complicate things.

For example, from Ivan Biuklija:
> For weeks, I've used a custom build of 2.1 with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN define without a mechanical Z-stop and everything worked fine. The only issue was the homing bug after cancelled prints, which is not related to this. I also never had to level the bed, regardless how thick the glass plate was, so I'm likely going back to that setup. For me, the only benefit of using a mechanical switch is slightly faster Z homing and a menu-driven Z-offset setting, neither crucial features in my opinion. 
Once I calculate the exact Z-offset, there is no reason this would change and I can set it in EEPROM and forget it.

If you have a opinion on that matter, do not hesitate to comment here (but comments with **facts**, not just "I use to do this or that".

john-f-chamberlain, 2018-04-15:
I think this might be a good idea, specially for those of us using your BL Touch mount. This should allow us to use a fixed mount heated bed instead of the spring mounted one which might(?) result in more stability during printing.

This would also allow us to mount the bed as close to the Y carriage as possible resulting in more Z height which might be nice especially for those of us using a sheet of glass on top of our bed.

Martin-Dolan, 2018-04-18:
I think the biggest argument for not removing functionality of the Z endstop is that people who use this firmware who don't have a BL-Touch or don't want to make electrical hardware modifications to their printers will need a different version of the firmware?

I've made the modifications needed to run the BL-Touch from the onboard Z probe and pin on the EXT connector as documented so I don't see a reason to start re-purposing other connectors. 

If the Z endstop is re-purposed and the BL-Touch fails for some reason, then the printer is not functional until the BL-Touch is replaced.   I know that is true for any part of the printer however the BL-Touch is an additional part and additional point of failure. If it's wired up the proper way it is simple to just stop using it and continue on with the Z endstop switch as before.

biuklija, 2018-04-18:
andyfarthing, I believe what Sebastien meant as "complication" (and what I originally referred to in my suggestion) is the fact that when you have both the Z-stop and the BLTouch, you have to carefully level the bed to keep it in the right zone above the mechanical Z-stop because probing fails if the Z-stop is triggered. 
I use different glass plates (different surfaces and thicknesses) and this would require that I re-level the heatbed before using Z-probing, which in my case defeats the purpose. Additionally, Z-probing fails are currently quite problematic because of [issue 96](https://github.com/andrivet/ADVi3pp-Marlin/issues/96). 
Menu-assisted Z-probe offset setting could be accomplished without the mechanical Z-stop - simply Z-home the heatend, disable software endstops, give the user control over the Z-height (+/- buttons) and tell the user to lower the heatend to 0.1. Measure the number of steps and that's it - the same way one would do it via terminal. 
Martin-Dolan, users without BLTouch already need a different version of the firmware. 
I believe we were not debating re-purposing the Z-end switch connector, only using the BL-touch alone as the Z-end switch. Everything can remain wired as per Sebastien's instructions (apart from the mechanical Z-stop which you would disconnect/remove). If the probe fails, the user can just put the Z-switch back and use a standard version of the firmware.

andyfarthing, 2018-04-18:
@biuklija Okay, I can see why that needs looking at. I actually ran into that problem when I was setting it up in this new way, but I just thought I was being stupid and adjusted the endstop to compensate. I never ran into it again, but if you're changing the thickness of the plate regularly then I can see why that could get annoying.

> I believe we were not debating re-purposing the Z-end switch connector, only using the BL-touch alone as the Z-end switch

For some reason I was stuck in the mindset of the old wiring (from the other fork), but that's obviously not the case! If we're not touching the endstop wiring then that can be easily connected / disconnected if needed without opening the case, which is good.

andrivet, 2018-04-18:
As you said, everything can remain wired. It will just be a question of which pin is used in the configuration of Marlin. I have no plan to re-purposed the Z-endstop or its pins. There are enough usable pin with the EXT connector. And also, I need a way to test both firmware without rewiring anything.

Originally, I planned to have only 1 firmware for all sensors (including manual leveling/no sensor), but it is not easy with the current Marlin (several features relies on compilation flags to get a binary as small as possible). So for the moment, it is safer to generate a different binary for each kind of sensor.

Thanks for all your comments.

vertigo235, 2018-05-02:
I was experiencing an issue today where my z stop was triggering as the z was moving down during the first layer. So I definitely think it's a good idea to use the z probe to establish the z stop.  

I also think it would be best to use the center of the bed for the probe z stop home.

vertigo235, 2018-05-03:
Works as expected.

andrivet, 2019-05-26:
Implemented in version 4.0

closed 2019-05-26 by andrivet