andrivet/ADVi3pp #100: [FR] Choose optimal baudrate (i.e. 115200 vs 250000)

andrivet, 2018-04-16:
Today, the firmware uses a default baudrate of 115200 (for both the LCD serial and the USB). However, it is well know that this rate is not optimal for Atmel ATmega2560 (mainboard MCU). If you look at table 22-12, page 226 of [ATmega640 1280 1281 2560 2561 datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-2549-8-bit-AVR-Microcontroller-ATmega640-1280-1281-2560-2561_datasheet.pdf), there is an error of -3.5% at 115200 bps (for 16MHz). The error is 0 at 250000 bps. Several printers are using 250000 bps by default for this reason even if this baudrate (250000) is not standard. 

So why not doing the same for ADVi3++ if it can reduce transmission errors? A fork of ADVi3++ from @ SureshotM6 (https://github.com/SureshotM6/ADVi3pp-Marlin) already uses 250000 for both baudrates. I already made myself some tests (visible in several commits) but they were not exhaustive enough, especially between the LCD and the mainboard.

If you have any experience on this subject, do not hesitate to post a comment (but based on **facts**, not just "it's useless" as I already read on the Wanhao Facebook group).

Cyber-One, 2018-08-14:
I haven't tried this speed on the Atmega series of microcontrollers, and i have no doubt that it would work on those controller with no real trouble. 
I would mention though, when you download the 1.1.9 of Marlin, that is now the default baud rate set.

andrivet, 2018-08-14:
"I haven't tried" and "i have no doubt". These are not very strong arguments... ;-) There is a big difference between Arduino Mega 2560 and Wanhao i3 Plus mainboard: the USB chip. In the case of the Wanhao, it is a low cost, cheap and not so reliable chip (http://www.wch.cn/download/CH340DS1_PDF.html). Moreover, 250000 is not a standard baudrate. So I am not as sure as you that it works without problems.

spikespaz, 2018-12-28:
@andrivet When I try to use 250000 as my baudrate from Repetier Host and the firmware, Windows straight up crashes after the printer does some weird shit. Like really, it just retracted all of the filament, moved Y to home, then moved back and tried to continue printing... without the filament. Then my laptop crashed. Full green screen of death. But when I leave it at 115200, it just skips around and continues doing weird shit so I really don't know how to get it printing from USB.

andrivet, 2018-12-30:
@spikespaz I do not see how ADVi3++ running on the mainboard of the printer can make a green screen of death on your laptop. The problem is on your computer side (with your USB driver more precisely).

spikespaz, 2018-12-31:
@andrivet It's actually not that hard to BSOD a computer over a USB port. But you're right, in this case, it wasn't my computer either, it was Repetier Host not using ping-pong. When I raised baudrate to 250000 and turned on ping-pong communication, the printer and print host on my laptop started communicating synchronously and the problem disappeared. I think there is a bug with ADVi3++ that causes this issue when ping-pong communication is turned off, I haven't had this issue with either of my two previous 3D printers, all running Marlin.

Apologies for the profanity. I apparently wasn't thinking clearly? I usually don't swear on GitHub because it tarnishes my reputation as well.

andrivet, 2019-05-26:
After many tests, go back to 115200. It is more reliable. At 250000, I do see communication problems.

closed 2019-05-26 by andrivet