andrivet/ADVi3pp #108: [FR] Live Z adjusment

strongbow1, 2018-04-28:
------------------------------------------------------------
### Feature Request
Live Z-Adjustment would be Awesome to have when Printing with different Filaments

andrivet, 2018-04-28:
Can you please provide some information about what you have in your mind? Like a link to a similar feature.

strongbow1, 2018-04-28:
Prusa i3 MK2, MK2s and MK3 have this function. I believe if you use Marlin 1.1.8 under the configuration_adv.h tab “Babystepping” will activate this function. I have done this with one of my own builds but its not as complicated as with the Wanhao i3 plus

Mit Freundlichen Grüsse
With Best Regards
Sandro Strimer

andrivet, 2018-07-04:
It may look like this:

[image]

chschmit, 2018-07-05:
What does the icon on the right signify? It looks like a baby carriage.

andrivet, 2018-07-05:
@chschmit I let you guess. The answer is in the title of this issue

andrivet, 2018-08-18:
Note to myself: Pressing "Z" on the Main screen during a print should display the babystepping adjustment screen (ie print settings). Maybe, pressing on temps should to.

closed 2018-11-04 by andrivet