andrivet/ADVi3pp #111: Adding G29 in the Octoprint startup G code causes a probe tip crash

tobashadow, 2018-04-29:
Running Beta3, If you ad G28 and G29 into the startup GCODE inside of Octoprint to run before every print the printer will home then deploy the probe and crash the probe into the bed while going to the first position causing the probe to error. Beta2 acts normal.

Running G28 and G29 in terminal acts normal

tobashadow, 2018-04-30:
If sending manually it will crash also if you don't wait about 2 seconds. The probe deploys before it is in position.

vertigo235, 2018-05-02:
Try adding a M400 command between the G28 and the G29

bobbydamaker, 2018-05-06:
I can confirm seeing this behavior as well.  Have to do a restart to get normal behavior back.  I would note that I have gotten a successful print with Octoprint sending G28 followed by G29, but as long as it is the first time I have run the BLTouch after powering up.  Seems to be a problem after cancels or a print completes.

andrivet, 2018-05-06:
I am not able to reproduce this behaviour either in OctoPrint "Before Print" (you are talking about this, right?) or manually. Please provide more information, like your complete startup G-Code. Are you able to reproduce this consistently? if yes, can you please provide a precise list of actions step by step?

tobashadow, 2018-05-06:
Type G28 into terminal then just as it hit the corner type in G29 and it will drop the probe and run into the bed.

I can not try to replicate it anymore due to changing to center of bed probe homing.

bobbydamaker, 2018-05-06:
I actually added a M400 before the G28 and a second M400 before the G29 (they were sequential in my pre script after waiting for heating the bed).  Since doing this, I haven’t had the problem either.  I only had this happen with Octoprint initiating the print.  Direct or SD card did not happen.  I will post my script when I can get to my computer with Cura and login.

tobashadow, 2018-05-06:
It's a recent fork of yours that changes G28 to use the the probe in center of the bed instead of the Z stop to find bed height. Works really good and allows you to change surfaces including different thickness of surfaces just by running G28.

andrivet, 2018-05-06:
Closing for the moment, since it is not reproducible. Do not hesitate to re-open but, please, give enough information.

closed 2018-05-06 by andrivet

bobbydamaker, 2018-05-06:
This is what I have now that is working.  I agree, it probably can still be closed for now.  It may have been something in my old script that I was using with S3D that didn't play nice in Cura. 

G21 ; set units to millimeters
G90 ; use absolute positioning
M82 ; absolute extrusion mode
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
M400
G28 W ; home all
M400
G29 ; mesh bed leveling
G92 E0.0 ; reset extruder distance position
G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E21.5 F1000.0 ; intro line
G92 E0.0 ; reset extruder distance position