andrivet/ADVi3pp #116: When running G29 while the extruder is hot, the extruder temp drops

andrivet, 2018-05-06:
Reported by Zachary Cornelius on Patreon.

### Bug Report
- Description: When running a G29 bed level while the extruder is hot, the extruder temp drops (like the firmware just isn't paying attention to the hotend while G29 is running)
- Expected behaviour: Temperature remain the same
- Actual behaviour: Temperature drops
- Steps to reproduce:
  - G28
  - M109 S200
  - G29
- Note: After the G29 is completed, the temperature raises again until it reaches its target

miriad, 2018-05-07:
Wouldn't surprise me if this was to reduce CPU utilization, but it definitely seems broken. The first few seconds of printing are ~20 degrees too cold.

vertigo235, 2018-05-12:
This is a feature, by design it disables the hotend so the current doesn't interfere with the probe.

vertigo235, 2018-05-12:
In configuration.h

/**
 * Enable one or more of the following if probing seems unreliable.
 * Heaters and/or fans can be disabled during probing to minimize electrical
 * noise. A delay can also be added to allow noise and vibration to settle.
 * These options are most useful for the BLTouch probe, but may also improve
 * readings with inductive probes and piezo sensors.
 */
#define PROBING_HEATERS_OFF       // Turn heaters off when probing
#define PROBING_FANS_OFF          // Turn fans off when probing
//#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors

vertigo235, 2018-05-12:
I'm on an iPhone so I can't see the exact line number.

andrivet, 2018-05-12:
@vertigo235 You're perfectly right, lines 691+. Thank you so much. I will not spend hours searching for nothing...
So close because this intentional.

closed 2018-05-12 by andrivet