andrivet/ADVi3pp #119: [FR] Support for Wanhao Duplicator i3 Plus Mark II

fujin, 2018-05-10:
Apologies if this is being tracked elsewhere, I had a quick search but couldn't find anything.

I recently purchased a Di3+ which ended up being a Mk. II when it arrived 😁.

I note in the manual:
> Currently, ADVi3++ is only compatible with the Wanhao Duplicator i3 Plus version 1 printer and its clones. Version 2 (aka Wanhao Duplicator i3 Plus Mark II) will be supported in a future release.

If there's something I can do to help this along, let me know, otherwise feel free to close if invalid/tracked elsewhere or similar! Excited to try out `ADVi3++`!

churd83, 2018-05-11:
Hey fujin! Are you working on butterflight? I assume you are :) Anyways I am about to compile a version of this firmware for my Mark II and I'll let you know if it works. I have flashed a previous version and it took the flash and operated but didn't work with bed leveling so I edited His source to work with Inductive sensors and I'm about to test it. Hopefully the LCD firmware will work with it as I do not know what to do with that. We shall see. I'll report back

churd83, 2018-05-11:
Nope didn't work correctly, LCD buttons didn't all work, but i could control the printer by gcode commands through octoprint somewhat. Back to the drawing board :(.
I told the dev that the source code for the Wanhao D9 had if/else statements for the D9 and the i3 MarkII but he said he wouldn't look at it. I don't know much about coding but can understand most of it. If you want to go look at it, it's on github

andrivet, 2018-05-12:
@churd83 I said I will not look at it **for the moment** (i.e. not before final release of 3.0). And I said also I will need help to support Mark II.

andrivet, 2018-05-12:
@churd83 When you are talking about D9 source code, you are referring to this: https://github.com/garychen99/D9 ?

churd83, 2018-05-12:
@andrivet I'm not going to bicker about what you said because I appreciate your work even though I can't use your hard work. Yes that is the source I was talking about when I pointed it out to you on Facebook. I just wanted you to look at it to see if it would help you merge for the MarkII. Like I said I'm not a coder, nor pretend to be, just trying to help.

churd83, 2018-05-12:
@andrivit I'm willing to be a tester for the MarkII. Also @fujin does coding so he might be able to help since he has the MarkII as well

fujin, 2018-05-13:
@churd83 I work on Betaflight (ew, butterflight.. 🤢)

@andrivet that's understandable regarding not before 3.0. If there's something in particular I can help with regarding the Mark II support, I'm happy to do my best. I have some amount of familiarity with embedded sw/hw practices from my experience with flight controllers.

andrivet, 2018-05-13:
I will probably add Mark II support at the same time than proximity sensor support (#52) for Mark I.

andrivet, 2018-05-21:
I add here the pictures I have about the Mark II. Do not hesitate to add more (of the mainboard for example).

[image]
[image]
[image]
[image]
[image]
[image]
[image]

domverity, 2018-05-25:
+1 for mkII support, happy to offer testing, photos etc if it would be useful...

andrivet, 2018-05-27:
@all Do you receive a manual with the Mark II? If yes, can you post here the pages about bed leveling?

domverity, 2018-05-27:
The physical manual I received with mine was ancient (doesn't even mention auto bed levelling), the new one is here https://drive.google.com/file/d/1-WZi-uHYdpV6g0gGUegqxxxRJ-6XwiMW/view?usp=sharing though don't get too excited, it's not the most detailed...

fujin, 2018-05-27:
Confirming the manual did not have anything regarding auto bed levelling.

These links sent to me by Wanhao may be of assistance:
- https://www.dropbox.com/s/6xwsdjzv5ljfsg1/OQD08017.A.ABL-SENSOR%281%29.pdf?dl=0
- https://www.dropbox.com/s/pphdambavm4occf/OQD08021.doc%20Sensor%20replacement.pdf?dl=0
- https://www.youtube.com/watch?v=8U04vtcbisM&feature=youtu.be

churd83, 2018-05-28:
@avdrivet what exactly are you needing to know from the manual for the ABL sensor. I might know what you need to know. Why not just make a list of what you need to know and I'm sure me and others can get you that info.

andrivet, 2018-05-28:
@fujin I look at the video. The adjustment of the sensor is not easy.

fujin, 2018-05-29:
@andrivet agreed. The good news is, once set, it seems to stay put. I also believe it may be possible to use [M851](https://reprap.org/wiki/G-code#M851:_Set_Z-Probe_Offset) with the stock Marlin 1.1.7 `M851 Z-XX.XX` for fine grain adjustment (e.g. +/- 0.01) although I've not tested as of yet. It defaults to `Z-0.20`

I ordered a BLTouch just in case this ends up being difficult.

churd83, 2018-05-29:
@fujin yes you will need to use the m851 command to fine tune your z offset because everyone's sensor won't trigger at the same height, nor will everyone's sensor be mounted at the same height. @andrivet should State that in the write-up for this firmware/printer.

deloosem, 2018-06-05:
I can't wait for your support of the mark 2. As of now the wanhao i3 plus mark II autolevel does NOT work.
It even messes a perfect level up ... lol (but then again the supplied allen wrenches just snap in two by just looking at them :)  )

bgoldman69, 2018-06-07:
has the auto level fail been fixed yet?

andrivet, 2018-06-07:
@deloosem I will implement Mark II support only if supporters on Patreon votes for it (I see you will be able to vote since you are a supporter of the project).

@bgoldman69 "has the auto level fail been fixed yet?". I do not know what you are talking about. And for support, please use Facebook ADVi3++ page or Patreon. GitHub issues are reserved for technical things.

deloosem, 2018-06-07:
Where do I vote?

andrivet, 2018-06-07:
Votes are not yet open. See: https://www.patreon.com/posts/on-road-to-4-19201225

Electomanic, 2018-09-16:
Found som images on the PCB online and made a comparison between the plus and plus mark ii.
Looks fairly similar.

I don't have the printer yet but have ordered it and hope to get hacking once it arrives.

[image]

churd83, 2018-09-20:
If anyone on here is on the Wanhao Duplicator I3 facebook group, https://www.facebook.com/groups/i3plus/, I have ported Sebastian's hard work to work with the MarkII printer. There is a long write up on the Group if you are interested.

churd83, 2019-02-04:
here are some more photos
[image]
[image]

ringerc, 2019-02-17:
Hi folks. I have a mkII (well, "Cocoon Create Touch 3D " but it appears to be a Wanhao Duplicator i3+ mkII with a different decal and LCD bootscreen).

I'm an experienced Linux (C) developer with some limited embedded experience and I'm also not totally incompetent with Arduino. I'm totally new to the 3D printing world though I have some limited idea about CnC in general. I'm interested in getting involved here and helping out.

@andrivet, you said you needed help to get the MKII working with ADVi++ . What help do you need? Do you have any sort of project outline, working document, set of issues, or skeleton to work from? Or are you not at that stage yet and more at the "figure out what's different and why it doesn't work" phase?

Either way I'm keen to get involved, though I do have extremely limited time thanks to work and young kids.

But the stock UI on this thing is unpleasant to use, the control firmware is clearly limited, and I'm keen to get my hands dirty.

You can ping me on

        python3 -c 'import base64; print(base64.b64decode("cmluZ2VyY0ByaW5nZXJjLmlkLmF1"))'

by email or gchat.

ringerc, 2019-02-17:
@churd83 Would you mind adding a link to your port/fork here?

andrivet, 2019-02-17:
@ringerc I have already merged the work of @churd83. His source code is in his github repository

andrivet, 2019-02-17:
@ringerc The only think I need is tests and to know if you have modified your printer and how.

ringerc, 2019-02-17:
@andrivet I was mistaken - it's a mk1 clone but the vendor has taken a selection of the more minor enhancements from the mk2: improved thumbwheels for bed leveling, improved power connection so there's no need for the MOSFET hack, etc.

No auto bed leveling, etc. Sorry for wasting your time.

andrivet, 2019-05-26:
Supported in version 4.0 except the power loss recovery (https://github.com/andrivet/ADVi3pp-Marlin/issues/205)

closed 2019-05-26 by andrivet