andrivet/ADVi3pp #123: Stuck on Load filament screen

dascoadams, 2018-05-15:
Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: ---stuck on load filament screen apron startup
- Expected behaviour: ---load up the AVD13 menu
- Actual behaviour: ---freezes on load filament
- Steps to reproduce:turn on
  - Do this
  - Do that

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

andrivet, 2018-05-15:
Please provide the required information. "Stuck on Load filament screen" is not a bug report.

dascoadams, 2018-05-15:
Where do I report it then

andrivet, 2018-05-15:
Here, with the required information (follow the template please)

dascoadams, 2018-05-15:
I hate to look so stupid ..but this is the second time I have ever posted here and I really don't understand the template. I tell everyone about your firmware because it is so easy. I'm old and learning

andrivet, 2018-05-15:
Close because required information not provided. If you are not telling me how to reproduce the bug, it can’t reproduce it and fix it.

closed 2018-05-15 by andrivet

dascoadams, 2018-05-15:
So the answer is to close the issue instead of helping me understand it. I can't tell you how to get your printer to freeze. If I could I would...all you do is turn it on and its frozen. There is now other way to my knowledge.

andrivet, 2018-05-15:
I am closing because I can’t do anything with so few information. This is not the right place to ask for support. This place is ONLY to report bugs (by filling properly the template) or requesting new features. If you need support, there are pages on Facebook and a project on Patreon.

dascoadams, 2018-05-15:
I really like what you have done. I will pay to understand it better. I have asked on Facebook with your name in the main post I have messaged you before and I get it. Your a very busy person. I ruined my motherboard trying to solve this issue on my own. Does patreon let me talk directly to you without wasting space here?

andrivet, 2018-05-16:
https://www.facebook.com/advi3pp/

andrivet, 2018-05-18:
I have no news from you. Or am I wrong?