andrivet/ADVi3pp #125: [FR]  Ability to enter Touch Sensor (X, Y) offset position by LCD instead of hard coded

Martin-Dolan, 2018-05-21:
Currently the (X, Y) position of the touch sensor probe in relation to the hotend nozzle is hardcoded. It would be great if it could be entered via the LCD instead of having to recompile the firmware.

Perhaps an option on the screen where you set up the sensor to enter  the offset of -15mm to +15 mm on the X and Y axis?

Or a screen where the nozzle moves to the middle of the bed and waits, (with the Z height at say 10mm). 
You mark on the bed under the nozzle with a sharpie.
You then use X, Y arrows to move the touch sensor over the sharpie mark and when correct press conform button.
The offset is calculated and stored in non volatile memory.

lightseeker34, 2018-05-30:
I have designed a custom toolhead and am dealing with this exact problem right now. To be able to do this without editing code would be a real game changer.

andrivet, 2018-08-18:
It may look like this:

[image]

andrivet, 2018-08-18:
Note to myself: In the User Manual, provide values (X, Y, Z) for my own supports but also for Teaching Tech ones (https://www.youtube.com/watch?v=BV11-VOQjMc)

andrivet, 2018-12-10:
Full implemented in Alpha 2

closed 2018-12-10 by andrivet