andrivet/ADVi3pp #126: ABL Stops at random heights with same g-code (Potential memory issue)

vertigo235, 2018-05-23:
Thank you for submitting your feedback to the ADVi3++ project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Bug Report
- Description: Auto Bed Leveling stops at unpredictable height levels, leaving one side squished and the other side of the print with a slight gap (non bonded). 
- Expected behaviour: Auto Bed Leveling continues adjustments throughout the print if fade out is disabled, or until fade out is completed. 
- Actual behaviour: It stop working completely after random print times. 
- Steps to reproduce:
  - Make sure the bed isn't level before printing, one side higher than the other. Load a tall wide print and print it.

Unfortunetly, it doesn't happen every time, even with the same G-code.  I was thinking that it might be the large mesh created by the bilinear subdivisions, causing too many calculations with certain meshes and then the ABL basically just stops or causes some sort of memory overflow.

I have my own custom build of the firmware so I disabled the Bilinear subdivisions altogether and I have not seen the issue again yet.  I know this could be coincidence and I did not do any debugging output to confirm this, but I wanted to post the issue here to see if anyone else was seeing the issue. 

I suspect that instead of completely disabling the bi-linear subdivisions, it could be decreased to 2 or maybe just 1 subdivision, however I'm printing on glass and I don't know if any subdivisions really help anyhow. 

------------------------------------------------------------
### Bug Report Tips
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

miriad, 2018-06-02:
One thing I've found, particularly for left to right issues like this, is that any bend in the X axis (really the whole gantry) can yield one side too low, and the other too high. Perhaps this could be your gantry shifting during a print? I saw some similar things, and the z-braces fixed then once I got them adjusted

vertigo235, 2018-06-02:
No it's definitely the abl turning off mid print. Because I've seen the z axis before and after the issue and it's adjusting before but them no longer after. 

It might be related to the improper z jerk settings that were fixed recently but I have not seen the issue since I disabled the bilinear subdivisions. I've so just tried to keep my bed more level and I think that helps.

andrivet, 2018-06-02:
Subdivision is experimental in Marlin. So maybe there are some bugs there. I think I will disable it in the next version.

andrivet, 2018-06-06:
So for version 3.0.2, I disable ABL_BILINEAR_SUBDIVISION. I may reintroduce it when Marlin 1.1.9 is released. Close this issue for the moment. Will reopen if the problem persists

closed 2018-06-06 by andrivet