andrivet/ADVi3pp #130: Add a option to test the automatic leveling with G26

andrivet, 2018-06-02:
- [X] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Feature Request

- [X] Your request is not specific to your setup but can benefit to many people
- [X] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [X] You have provided a short but meaningful title
- [X] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [X] You have provided objective **facts**, not your personal feelings, your personal preferences.

Add a button (probably in Tuning / Sensor) to test the leveling with G26: http://marlinfw.org/docs/gcode/G026.html

The screen may includes options such as the temperature and layer height

andrivet, 2018-09-01:
@luigi311 "...remove our gcode as a point of failure.." I do not understand what it means. Can you clarify?

luigi311, 2018-09-01:
Sure, having the g26 command will print out a single layer validating the mesh. Having this function as a build in feature will remove the need for people to slice our own stl to create the gcode. This will have us use a known gcode script that will for sure use compensation if it has the ability too, right now the Facebook community has a problem where compensation is not activating and we are unsure if it is something with the gcode we are slicing up or not. With this we won't have to worry about the gcode we created to validate it and we will know if it's a problem with the gcode we sliced or not.