andrivet/ADVi3pp #139: [Fr] M600 Button on printing screen

LotDev, 2018-06-07:
I am sure that more than just I can benefit from this feature if you decided to implement it, and we will be thankful if you do. I know you been implementing the M600 command in the printer and it is quite useful but the problem is that it can be only used when connected through a computer (octoprint) or in the gcode. I think it will be a great thing if you had a button the preformed the M600 command on the printing page. Thanks for the time and your support and for your work on this project!

andrivet, 2018-06-07:
Can you please use the issue template?

LotDev, 2018-06-08:
Sorry about that, fixed it and used the template you provided!

andrivet, 2018-07-04:
It may look like this:

[image]

LotDev, 2018-07-06:
Looks great! Thanks for your work man!

noel2763, 2018-09-04:
Awesome. Thanks a lot. Will the LCD show Resume or Extrude more plastic button using M600 advance pause feature?

andrivet, 2019-05-26:
Implemented in version 4.0

closed 2019-05-26 by andrivet