andrivet/ADVi3pp #140: The automatic bed leveling detects, in some cases, the left side lower than the right side

andrivet, 2018-06-08:
Thank you for submitting your feedback to the ADVi3++ project.

- [X] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Bug Report

- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) issue (use the search)
- [x] You have tested with the **latest** version of ADVi3++ (including Beta releases)
- [x] You are able to reproduce the bug
- [x] You are able to describe step by step the process to reproduce the bug

* Exact model of your **printer**: Monoprice Maker Select Plus
* **ADVi3++** exact version: 3.0.1
* Printing using **SD card** or **USB**: n/a
* **For USB printing only** Software used to print with its exact version: n/a

#### Description of the bug

* Expected behaviour: When the bed is leveled (manually), the automatic bed leveling must produce values very close to 0. The standard deviation of BLTouch (from its datasheet is 0.01mm)
* Actual behaviour: The automatic bed leveling detects that the left side is lower than the right side.
* Steps to reproduce:
  - Carefully manually level the bed
  - Perform an automatic bed leveling.
  - The values in the grid are not close to 0 except in the middle of the bed

andrivet, 2018-06-10:
I have very carefully level manually the bed and then I have performed an automated bed leveling. Here what I get:

[image]

For me, the ABL works nicely. I have also checked than nothing is inverted. If I move a corner, it appears in the grid.

Dazzler24, 2018-06-13:
I'm experiencing a similar issue to the original writer.  Bed has been checked and confirmed as 'flat' using a straight edge.
I have carefully manually leveled the bed before using ABL.
The left side always reports as lower than the other six positions.
The first (quick) Z probe readings, as displayed on the screen when auto leveling, all appear to be approx 0.2 less on the left side when compared to the other six readings. 
ADVi3++ 3.0.2 - Cocoon Create Touch.
[image]

This is a typical ABL result.

Dazzler24, 2018-06-17:
Just wanted to mention that I was getting the same 'problem' as the 'bug' reported but have since discovered that my x axis rods were not perpendicular to the Z axis at one end only.  It was only slightly out compared to the opposite side but enough to make it appear as though one side was lower than the other when doing an ABL.  While a manual level could be achieved at the nozzle, because the sensor is some 50mm out to the front of the nozzle the angle created by the non aligned rods (one end) were magnifying the error in the ABL readings which then led to bed adhesion problems.

andrivet, 2018-06-17:
@Dazzler24 Very good explanation, thanks. I will mention this on the Patreon page. I will also try to find a way to minimize, or measure this effect.

Dazzler24, 2018-06-19:
@andrivet Sebastien, Thinking about the problem being an issue in the vertical plane, I was wondering if this problem could be minimised by having a bracket that placed the sensor in alignment with the nozzle in that plane. i.e. bring it around to the right hand side (fan being on the left and therefor in the way).  This solution would at least minimise (even negate!) any backwards/forwards tilt of the x axis rods while still compensating for any left/right tilt.  What do you think?

andrivet, 2018-07-15:
@Cyber-One "the grid it is testing is not aligned with the bed". This is normal and correct. The mesh takes into account that. Most of the time, the problem you have is because the printer is not square enough (X axis versus Z axis in particular).

Cyber-One, 2018-07-16:
Hello Andrivet,
That was also one of my first thought, but to be sure, I re-checked the level of the X-Axis carriage using y calipers.
There was yet again a 0.2mm error, however after correcting this to within 0.03mm the issue still persists.
[image]

I believe the issue is the high spot on y bed is in the exact center of the bed, with the test point on the right side and middle being either side of this, the esh is seeing the error as a slope in the bed or a poorly leveled bed when in fact it is not.
If the test points were more centered and potentially an option to increase the number of test points from a 3x3 grid to a 5x5 grid (this increase would be a feature upgrade) this type of error could be picked up in mesh leveling.

Current leveling is based on the following settings
  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 20
  #define RIGHT_PROBE_BED_POSITION 160
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 150

with the Pobe settings as
#define X_PROBE_OFFSET_FROM_EXTRUDER -29   // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -40   // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -1.54 // Z offset: -below +above  [the nozzle]

if we instead define the following settings
  // Set the boundaries for probing (where the probe can reach).
  #define PROBE_EDGE_CLEARANCE 10
  #define LEFT_PROBE_BED_POSITION (PROBE_EDGE_CLEARANCE + X_PROBE_OFFSET_FROM_EXTRUDER)
  #define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - PROBE_EDGE_CLEARANCE + X_PROBE_OFFSET_FROM_EXTRUDER )
  #define FRONT_PROBE_BED_POSITION (PROBE_EDGE_CLEARANCE - Y_PROBE_OFFSET_FROM_EXTRUDER )
  #define BACK_PROBE_BED_POSITION (Y_BED_SIZE - PROBE_EDGE_CLEARANCE + Y_PROBE_OFFSET_FROM_EXTRUDER )

This should assuming I got the +/- around the correct way center the probing on the bed.

The other method is to just work out more centralised probe point and hard code it... ;-)
I always like simple settings for users..

andrivet, 2018-07-19:
@Cyber-One The firmware does know where it probes and takes into account the exact position to construct the mesh. So I do not understand your explanation. Your suggestion of modifications is based on tests or only on guessing ("I believe")? Is it valid only in your specific case ("my bed")?

Cyber-One, 2018-07-21:
In an effort to explain what I was finding, I yet again:
1) checked the level of the X-axis, still within 0.05mm
2) Manually leveled the bed using the thumb screws, (the manual level menu makes this so much easier then it used to be)
3) Re-calibrated the Z-probe -1.35mm (which does occur in the middle of the bed, had to lower all 4 corners by 0.15mm. I have graduated thumb nuts to assist in this)
4) Run the Auto bed leveling again.
[image]

I saved this result and then started a print not using the mesh leveling to check the results.
[image]

With all this info, I created a mesh in a spread sheet and entered the data at the probe points as returned by the BLtouch auto bed level.

Using a linear approach, the mesh does indicate the middle at 0.16, but it also shows the right side being much higher than it is.

I will admit that my theory of miss sampled high spot is incorrect with further investigation, but i am now left with no idea on where to investigate this problem futher....

Why would the bed be showing such a dramatic rise on the right side when in practice this is not the case?

closed 2018-08-12 by andrivet