andrivet/ADVi3pp #143: Support for filament width sensor and volumetric extrusion

hnzxdv, 2018-06-11:
Integrating support for an optical filament width sensor like the one linked below would enable leveraging of Marlin's existing volumetric extrusion feature. This would allow the printer to dynamically change the extrusion multiplier in response to fluctuations in filament width. As a bonus, such a sensor could also be used to detect filament run-outs.

Some benefits would include:
-Greater extrusion accuracy
-Increased compatibility with homemade filaments or those with poor manufacturing tolerances
-Obviate the need to manually tune extrusion multipliers

Links:
Thingiverse project: https://www.thingiverse.com/thing:454584
Sensor: https://objectswithintelligence.weebly.com/store.html

andrivet, 2018-06-11:
Next time, please the template otherwise I will close the issue.

numerocasuale, 2020-05-03:
Is there also another version of the sensor? Like an honest price version for example? Or a non thief version? Asking for a friend.

andrivet, 2020-08-04:
After studying your feature request, it looks like very specific to the projects you mentioned. And apparently, the sensor is no more available for purchase (I tried). So unless there is a viable alternative, I will not implement this feature.

closed 2021-10-04 by andrivet