andrivet/ADVi3pp #144: Optional XY Homing Following M600

zjed42, 2018-06-12:
### Feature Request

Detailed description of the feature request:
I think it would be helpful to have the option of performing an X/Y Home immediately following an M600 Filament change. On several occasions I have accidentally moved the extruder while feeding in new stock, being able to re-home would potentially have saved the print. Note: This should be optional (Via Front Panel Button) and not part of the M600 routine. I would suggest a check box that, if enabled, would perform a G28 X Y immediately following the completion of M600.

andrivet, 2018-06-15:
Can you elaborate. I do not understand your point. There is alreday a way of triggering a x/y home from the LCD panel

zjed42, 2018-06-16:
I tested today and have two observations, the first is that when M600 is issued, the control panel is locked in to the change filament operation, as such there's no way to access the home functions until after the M600 operation completes. By this point it has already started printing again, and if a re-home was needed, it's too late to do so. My original suggestion was to allow the option to automatically insert a G28 X Y after the M600 completes, but before the print starts. 

The second observation is that with the stepper motors engaged, it seems a lot harder to move the extruder than I originally thought. With that said, it still possible, but perhaps less likely that a re-home will be necessary.

andrivet, 2018-06-16:
I better understand. Why not adding a home button (or buttons) on the change filament screen(s)?

andrivet, 2018-06-17:
I am just performing a print with several M600. Frankly, I do not understand the problem you want to address with a homing. The head never moved while changing filament. And what homing would change? I do not see the problem here. As a side note, G28 would be wrong in this context since it would disable ABL.

andrivet, 2019-05-26:
No feedback so closing

closed 2019-05-26 by andrivet