andrivet/ADVi3pp #147: Create a tool to help flashing ADVi3++: ADVi3++ Flasher

andrivet, 2018-06-16:
- [x] This issue is **NOT** about support. This is not the right place for support (there is an [ADVI3++ Facebook page](https://www.facebook.com/advi3pp/) for that)

------------------------------------------------------------
### Feature Request

- [x] Your request is not specific to your setup but can benefit to many people
- [x] You have checked that you are not creating a **duplicate** of an existing (maybe closed) request (use the search)
- [x] You have provided a short but meaningful title
- [x] You have provided enough information (such as links to web sites, blogs, technical description) so that anyone can understand what you are talking about. Just any only a title is not enough.
- [x] You have provided objective **facts**, not your personal feelings, your personal preferences.

Detailed description of the feature request:

Today, some people are not comfortable with the flashing process. They do not always know which release to flash, they do not understand that there are two parts and that these parts have to be at the same version level, they are not comfortable with the necessity to open the printer, etc.

So the goal if this enhancement is to provide as much help as possible. The requirements are more or less the following:

* Standalone, desktop tool. No need to install any external tools (i.e. no need to install Arduino IDE, ...)
* Multi-platform (macOS, Windows, Linux)
* Retrieve and display the list of available releases (version number, flavor such *as with BLTouch support*, perhaps the changelog)
* Check USB connectivity with the printer, the baudrate, any missing driver, etc. Help solve connectivity problems
* Retrieve the versions (both parts) of the firmware currently installed
* Flash the mainboard part with some progress
* If technically possible: flash the LCD part with some progress. I know there are technical solutions to flash the LCD without requiring the opening of the printer. However, the flashing process with way is very slow so I am not sure if it is actually viable.

jemhirch, 2020-03-10:
how do you get the hex file no longer available on your site

bleughb, 2020-03-10:
I’d hazard a guess

Pre-build binaries are published on ADVi3++ Community
<https://community.advi3pp.com/t/advi3-version-4-0-6/> but are only
available to sponsors of the ADVi3++ project on Patreon
<https://www.patreon.com/andrivet>. So if you are not a sponsor, you have
to generate the binaries yourself (including firmware hex files, icons,
fonts, images, etc). I will not give help to non-sponsors on that subject.

andrivet, 2020-03-10:
@bleughb This is not the right place to ask questions (ie. ask for support) and your questions have nothing to do with the topic your are posting under. Please stop.

jemhirch, 2020-03-10:
I am a patron supporter

andrivet, 2020-03-10:
@jemhirch so ask your questions on ADVi3++ Community, not here.